tango-controls/cppTango #535: Travis build fails on cppzmq for CentOS7

daneos, 2019-02-13:
I'm trying to set up Travis build for CentOS 7, however I got stuck on the `install_cppzmq.sh` script.
It's giving me these errors:
```
$ .travis/install_cppzmq.sh
Build cppzmq
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning at CMakeLists.txt:4 (find_package):
  By not providing "FindZeroMQ.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ZeroMQ", but
  CMake did not find one.
  Could not find a package configuration file provided by "ZeroMQ" with any
  of the following names:
    ZeroMQConfig.cmake
    zeromq-config.cmake
  Add the installation prefix of "ZeroMQ" to CMAKE_PREFIX_PATH or set
  "ZeroMQ_DIR" to a directory containing one of the above files.  If "ZeroMQ"
  provides a separate development package or SDK, be sure it has been
  installed.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.27.1") 
CMake Error at CMakeLists.txt:12 (message):
  ZeroMQ was NOT found!
-- Configuring incomplete, errors occurred!
See also "/home/tango/cppzmq/build/CMakeFiles/CMakeOutput.log".
The command ".travis/install_cppzmq.sh" failed and exited with 255 during .
Your build has been stopped.
```
I can see it's missing some cmake files, but where should I get those from? And why it's working for Debian builds?

You can see the configuration I have right now [here](https://github.com/synchrotron-solaris/cppTango/tree/centos7_build) (`centos7_build` branch of synchrotron-solaris' fork).

bourtemb, 2019-02-18:
Hi,

It looks like you are already on a good track to fix this issue.
This seems to be a cppzmq specific issue.

Your problem might have been fixed in cppzmq v4.3.0.
You could try to update travis.yml in order to retrieve cppzmq 4.3.0 instead of v4.2.2...

Your problem might still be there if centos7 is not providing libzmq static library.
In this case, this seems to be fixed in https://github.com/zeromq/cppzmq/commit/bffb0eda34195339f2af49f2879d9d0b9dee5c28 (cppzmq master branch) but this is not yet available in an official cppzmq release...
So you could try with cppzmq master branch first to see if this fixes your issue.

t-b, 2019-06-05:
@daneos Could you try to do your changes on top of #561? This PR should fix the version dependency on cppzmq.

PR #561 (2019-08-17): Be compatible with all in-use cppzmq versions.
Follow up from #530.

Close #499.
Close #273.
Close #535.
closed 2019-08-17 by t-b