tango-controls/cppTango #536: TRANSIENT_CallTimedout and 'CORBA::TRANSIENT'  process

jkotan, 2019-02-14:
Hello,

I would like to ask if it is allowed to use cpptango with processes (e.g. Starter server do it) and what are restrictions?
If I write a simple stress test with processes
```c++
#include <tango.h>
#include <stdlib.h>  
#include <sys/wait.h>
#include <sys/time.h>

void runworkers(){
  pid_t pids[10];
  int i;
  int n = 10;
  
  for (i = 0; i < n; ++i) {
    if ((pids[i] = fork()) < 0) {
      perror("fork");
      abort();
    } else if (pids[i] == 0) {
      
      struct timeval stime;
      gettimeofday(&stime, NULL);
      struct timeval etime;
      gettimeofday(&etime, NULL);
      double dtime = (double)(etime.tv_sec - stime.tv_sec)
	+ (double)(etime.tv_usec - stime.tv_usec) * 0.000001;
      while(dtime < 1.0){
	try{
	  Tango::DeviceProxy * dp = new Tango::DeviceProxy("test/tserver/1");
	  Tango::DeviceAttribute vl;
	  vl = dp -> read_attribute("Value");
	} catch(Tango::DevFailed &e){
	  
	  Tango::Except::print_exception(e);
	  
	}
	catch(std::string &e){
	  std::cerr << "ERROR: " << e << std::endl;
	}
	gettimeofday(&etime, NULL);
	dtime = (double)(etime.tv_sec - stime.tv_sec)
	  + (double)(etime.tv_usec - stime.tv_usec) * 0.000001;
      }
      exit(0);
    }
  }
  
  int status;
  pid_t pid;
  while (n > 0) {
    pid = wait(&status);
    --n;  
  }
}
  
  
int main(int argc, char ** argv){

  try{
    Tango::Database * db = new Tango::Database();
    Tango::DeviceProxy * dp = new Tango::DeviceProxy("test/tserver/1");
    for (int i=0; i<4; i++)
      runworkers();
  } catch(Tango::DevFailed &e){
    
    Tango::Except::print_exception(e);
    
  }
  catch(std::string &e){
    std::cerr << "ERROR: " << e << std::endl;
    
  }
  
}
  
```
I get a lot of errors like
```bash
Tango exception
Severity = ERROR 
Error reason = API_AttributeFailed
Desc : Failed to read_attribute on device test/tserver/1, attribute Value
Origin : DeviceProxy::read_attribute()

Tango exception
Severity = ERROR 
Error reason = API_CorbaException
Desc : TRANSIENT CORBA system exception: TRANSIENT_CallTimedout
Origin : DeviceProxy:read_attribute

Tango exception
Severity = ERROR 
Error reason = API_DeviceTimedOut
Desc : Timeout (3000 mS) exceeded on device test/tserver/1
Origin : DeviceProxy:read_attribute
```
However, when I comment
```c++
Tango::DeviceProxy * dp = new Tango::DeviceProxy("test/tserver/1");
```
in `main()` I get another errors
```bash
terminate called after throwing an instance of 'CORBA::TRANSIENT'
```
Futhermore, if I comment both
```c++
Tango::Database * db = new Tango::Database();
Tango::DeviceProxy * dp = new Tango::DeviceProxy("test/tserver/1");
```
the script runs without any error.
I've performed my tests in debian 10: tango-* (9.2.5a).
The errors do not depend on a tango server.

Similar issue can be observered for pytango https://github.com/tango-controls/pytango/issues/259

Does it mean if I use processes I'm not allowed to create DeviceProxy/Database objects in the main script (so all my tasks with DeviceProxy I have to delegate to my processes) or it is just a bug?

Bests,
Jan

jkotan, 2019-02-15:
Probably, to cure this situation it would be enough during creating Tango objects (Database, DeviceProxy, AttributeProxy, ...) compare the current pid with the one used during previous CORBA initialization (if it has happened) and if they are different initialize CORBA from scratch.

jkotan, 2019-02-15:
Other option would be to provide a command which reinitializes CORBA from scratch. Then in each new process  a user can call it. 
Does such a command exist?

bourtemb, 2019-02-18:
> Other option would be to provide a command which reinitializes CORBA from scratch. Then in each new process a user can call it.
> Does such a command exist?

Yes, it does exist!
In your example, if you invoke:

`Tango::ApiUtil::cleanup();`

at the beginning of each forked process it should work.

Please note that you have memory leaks in your example (The DeviceProxy objects are never deleted).
Do you want to stress the TangoDatabase server by creating DeviceProxy objects inside loops?
If you just want to stress the remote device server, you should reuse the DeviceProxy objects.

Hoping this helps.

jkotan, 2019-02-18:
Great. Thanks @bourtemb !  `Tango::ApiUtil::cleanup();` helps in the case.

closed 2019-02-18 by jkotan