tango-controls/cppTango #538: Cannot poll local dynamic command

bourtemb, 2019-02-20:
As reported in https://github.com/tango-controls/pytango/issues/256 and on the [Tango forum](www.tango-controls.org/community/forum/post/3845/), there can be a problem to start the polling on dynamic commands.

This problem can occur when the command is added with _add_command()_ method and when the _device_ boolean parameter passed to _add_command()_ is set to true, which means the command is added only for this device.
The problem does not occur if the _device_ parameter is set to false.

When the _device_ parameter is set to false, it seems the polling on the dynamic commands is not automatically started if the polling period was defined from POGO. This is not the expected behaviour.

mliszcz, 2019-12-27:
Hi @bourtemb 

In polling configuration, we ignore entries fom `DeviceImpl::command_list` (the one returned by `get_local_command_list`). We also need to try to start polling after the user calls `DeviceImpl::add_command`.

We may address this issue in two ways:

Either we go through the whole codebase and fix all / most calls to `DeviceClass::get_command_list` / `DeviceClass::get_cmd_by_name` to also include commands from `DeviceImpl::command_list`. This requires some effort and analysis, e.g. whether it is ok to have dynamic commands in Device or it is only for Device_2 (and latter).

Or we may change just enough code to resolve this issue and address any future issues in dedicated PRs.

I'm in favor of option 1 - unify class and device commands when possible. e.g. in DeviceImpl we may have `get_all_commands` method which returns combined list of all (class + device) commands. Or `get_cmd_by_name`, which will first look for a class command and then for a device command. We will use these new accessors in place of current ones.

Below changes solved the issue for me:

1. command getter - quite inefficient but works as a **proof-of-concept
   ```cpp
   std::vector<Command*> DeviceImpl::get_all_commands()
   {
       auto& device = device_class->get_command_list();
       auto& locals = get_local_command_list();

       std::vector<Command*> all{};
       all.insert(all.end(), device.begin(), device.end());
       all.insert(all.end(), locals.begin(), locals.end());
       return all;
   }
   ```
1. Use new `get_all_commands` in below calls:
   https://github.com/tango-controls/cppTango/blob/4b280b46ca81e71abc84e1b6160efee3f1138bc0/cppapi/server/device.cpp#L817
   https://github.com/tango-controls/cppTango/blob/4b280b46ca81e71abc84e1b6160efee3f1138bc0/cppapi/server/device.cpp#L4258
   https://github.com/tango-controls/cppTango/blob/4b280b46ca81e71abc84e1b6160efee3f1138bc0/cppapi/server/dserverpoll.cpp#L268
   https://github.com/tango-controls/cppTango/blob/4b280b46ca81e71abc84e1b6160efee3f1138bc0/cppapi/server/utils_polling.cpp#L1440
   https://github.com/tango-controls/cppTango/blob/4b280b46ca81e71abc84e1b6160efee3f1138bc0/cppapi/server/dev_poll.cpp#L87
1. At the end of `DeviceImpl::add_command`, setup polling for new commands (if any):
   ```cpp
   init_cmd_poll_period();
   Tango::Util *tg = Tango::Util::instance();
   if (tg->_UseDb == false)
       init_poll_no_db();
   ```

mliszcz, 2020-02-03:
@bourtemb @t-b @Ingvord do you have any opinions? IMO we should unify access to commands, i.e. provide single iterator that iterates over all (class + device) commands and replace all command access across the whole codebase to use this iterator.

bourtemb, 2020-02-03:
> IMO we should unify access to commands, i.e. provide single iterator that iterates over all (class + device) commands and replace all command access across the whole codebase to use this iterator.

Your proposal sounds good to me.

closed 2022-02-24 by t-b