tango-controls/cppTango #540: segmentation fault reading attribute (or accessing DevProxy in general) during application shutdown

delleceste, 2019-03-08:
Hello.
This sample code represents a Qt class which is created at application startup and then destroyed when the application exists.

CuCustomWidgetCollectionInterface::CuCustomWidgetCollectionInterface(QObject *parent): QObject(parent)
{
    dev = new Tango::DeviceProxy("test/device/4");
    try {
        Tango::DeviceAttribute da;
        da = dev->read_attribute("double_spectrum_ro");
        std::vector<double> vd;
        da >> vd;
    }
    catch(Tango::DevFailed &e) {

    }
}

CuCustomWidgetCollectionInterface::~CuCustomWidgetCollectionInterface()
{
    printf("\e[1;31mo\e[0m ~CuCustomWidgetCollectionInterface %p\n", this);
    try {
        Tango::DeviceAttribute da;
        da = dev->read_attribute("double_spectrum_ro");  // SEGFAULT
        std::vector<double> vd;
        da >> vd;
    }
    catch(Tango::DevFailed &e) {
    }
}

This class is destroyed by Qt at the very end of the application life cycle. It is managed by Qt plugin system, so it is created and free by Qt.
In practise, I would like to accurately free Tango DeviceProxies when the Qt designer releases its resources at close time.

In normal Qt applications, when it's me who decides the proper point when to destroy Tango DeviceProxy references, I do not face the issue.

I guess there is a moment in an application life **after which** it becomes unsafe to access a DeviceProxy previously instantiated and used.

Can you confirm this?
What is this moment? (I remind from former releases something called the shutdown of the ORB)
Is it possible to fix it?

Thanks in advance.

Giacomo. ELETTRA

bourtemb, 2019-03-08:
Hi @delleceste ,

Indeed, if you access DeviceProxy objects after the ORB has been shut down, you will be in trouble.
Do you call 
`Tango::ApiUtil::cleanup();`
somewhere in your code?

Tango::ApiUtil::cleanup() will , among other thing, shutdown the ORB.
You should not access DeviceProxy objects after this call.

t-b, 2019-03-08:
@delleceste Could you use code blocks for the code?

@bourtemb 
But we could just throw in the `DeviceProxy` members if orb is a nullptr. It doesn't has to crash.

delleceste, 2019-03-08:
Hello Reynald. Thanks.
No, I don't explicitly call ApiUtil::cleanup anywhere from my code.

Giacomo

-- 
Giacomo S.
http://www.giacomos.it

Sincrotrone Trieste S.C.p.A. di interesse nazionale
Strada Statale 14 - km 163,5 in AREA Science Park
34149 Basovizza, Trieste ITALY
-
040 3758073
328 3237959

- - - - - - - - - - - - - - - - - - - - - -

* iqfire-wall, un progetto
  open source che implementa un
  filtro di pacchetti di rete per Linux,
  e` disponibile per il download qui:
  http://sourceforge.net/projects/ipfire-wall

* Informazioni e pagina web ufficiale:
  http://www.giacomos.it/iqfire/index.html

- - - - - - - - - - - - - - - - - - - - - -

 . ''  `.
:   :'    :
 `.  ` '
    `- Debian GNU/Linux -- The power of freedom
        http://www.debian.org

bourtemb, 2019-03-08:
Giacomo,

How is the Qt plugin system managing your classes?
Is it loading and unloading dynamic libraries, [it seems](https://doc.qt.io/qtcreator/adding-plugins.html)...

I got some questions recently from someone who was doing something similar on Windows and had trouble when unloading the library using Tango.

The solution was to call
`Tango::ApiUtil::cleanup();`
to shutdown the orb and release all the CORBA and ZMQ involved resources before unloading the library.

Shutting down the ORB before unloading a dynamic library using CORBA is actually what is recommended in [omniORB documentation](http://omniorb.sourceforge.net/omni42/omniORB.pdf) (section 2.8.6 ORB destruction, page 18).

Could you please try that if that fit your use case?
Cheers,
Reynald

bourtemb, 2019-03-08:
> @bourtemb
> But we could just throw in the `DeviceProxy` members if orb is a nullptr. It doesn't has to crash.

I made a simple test, trying to use an already created DeviceProxy after the ORB has been shut down.
We get a `BAD_INV_ORDER CORBA system exception: BAD_INV_ORDER_ORBHasShutdown` exception in this case, as in https://github.com/tango-controls/cppTango/issues/449

delleceste, 2019-03-08:
Thanks Reynald.
I didn't receive a BAD INV ORDER rather an segmentation fault pointing to a
mutex_unlock.
I can provide full backtrace monday.

I do not want to perform any action during library unload on the
DeviceProxy, but simply deleting it causes the same trouble.
So maybe the library is unloaded after ORB shutdown and DeviceProxy cannot
be deleted there.
So I see no way to work it around right now!
A trick is to get a reference to the plugin a little bit EARLIER In the
code and delete the device there.
But it would be good not to have this nuisance of course!

Thanks again.

Giacomo.

delleceste, 2019-03-11:
Hello Reynald.
This is the backtrace I get from gdb:

Core was generated by `./bin/cumparsita forms/spectrum.ui test/device/1
test/device/2'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fe59127bdf4 in pthread_mutex_lock () from
/usr/lib64/libpthread.so.0
[Current thread is 1 (LWP 19684)]
(gdb) bt
#0  0x00007fe59127bdf4 in pthread_mutex_lock () from
/usr/lib64/libpthread.so.0
#1  0x00007fe592a5192e in omni_mutex::lock (this=<optimized out>) at
../../../../include/omnithread.h:255
#2  omni::duplicateObjRef (objref=0xf5f130) at omniInternal.cc:305
#3  0x00007fe593475519 in Tango::Device_5::_duplicate (obj=0xf5f110) at
/home/giacomo/devel/cppTango/cppapi/server/idl/tangoSK.cpp:6080
#4  0x00007fe59339972a in Tango::DeviceProxy::read_attribute
(this=0xf4f790, attr_string=...)
    at /home/giacomo/devel/cppTango/cppapi/client/devapi_base.cpp:5919
#5  0x00007fe59333bd4f in Tango::DeviceProxy::read_attribute
(this=0xf4f790, att_name=<optimized out>)
    at /home/giacomo/devel/cppTango/cppapi/client/DeviceProxy.h:666
#6  0x00007fe58d00ea28 in
CuCustomWidgetCollectionInterface::~CuCustomWidgetCollectionInterface
(this=0xf35a70, __in_chrg=<optimized out>)
    at cucontrolsplugin.cpp:239
#7  0x00007fe58d00eaa1 in
CuCustomWidgetCollectionInterface::~CuCustomWidgetCollectionInterface
(this=0xf35a70, __in_chrg=<optimized out>)
    at cucontrolsplugin.cpp:232
#8  0x00007fe59158cebb in ?? () from /usr/lib64/libQt5Core.so.5
#9  0x00007fe59158e6b5 in ?? () from /usr/lib64/libQt5Core.so.5
#10 0x00007fe590dbc2a7 in __cxa_finalize () from /usr/lib64/libc.so.6
#11 0x00007fe5913b27b3 in ?? () from /usr/lib64/libQt5Core.so.5
#12 0x00007ffd3f418140 in ?? ()
#13 0x00007fe593d3ff76 in _dl_fini () at dl-fini.c:138
Backtrace stopped: frame did not save the PC
(gdb)

Best regards

Giacomo.

-- 
Giacomo S.
http://www.giacomos.it

Sincrotrone Trieste S.C.p.A. di interesse nazionale
Strada Statale 14 - km 163,5 in AREA Science Park
34149 Basovizza, Trieste ITALY
-
040 3758073
328 3237959

- - - - - - - - - - - - - - - - - - - - - -

* iqfire-wall, un progetto
  open source che implementa un
  filtro di pacchetti di rete per Linux,
  e` disponibile per il download qui:
  http://sourceforge.net/projects/ipfire-wall

* Informazioni e pagina web ufficiale:
  http://www.giacomos.it/iqfire/index.html

- - - - - - - - - - - - - - - - - - - - - -

 . ''  `.
:   :'    :
 `.  ` '
    `- Debian GNU/Linux -- The power of freedom
        http://www.debian.org

bourtemb, 2019-03-11:
@delleceste 
Thanks for the backtrace.
What version of cppTango are you using?

I tried to delete a DeviceProxy object after the ORB had been shutdown. No crash, no error.

Could it be that Qt is unloading the Tango library before your library?
Do you load several similar Tango plugins together?

delleceste, 2019-03-12:
Hello Reynald

> @delleceste <https://github.com/delleceste>
> Thanks for the backtrace.
> What version of cppTango are you using
freshly cloned end of last week

> I tried to delete a DeviceProxy object after the ORB had been shutdown. No
> crash, no error.
>
> Could it be that Qt is unloading the Tango library before your library?

That's quite likely. The workaround I described in one of my first emails
deletes device proxy earlier in order to avoid the crash.
But how would I know exactly when and how Qt unloads Tango library?

> Do you load several similar Tango plugins together?

Actually maybe the QTango alongside the cumbia one.
I could try with one only if you think that makes a difference.

Giacomo

-- 
Giacomo S.
http://www.giacomos.it

Sincrotrone Trieste S.C.p.A. di interesse nazionale
Strada Statale 14 - km 163,5 in AREA Science Park
34149 Basovizza, Trieste ITALY
-
040 3758073
328 3237959

- - - - - - - - - - - - - - - - - - - - - -

* iqfire-wall, un progetto
  open source che implementa un
  filtro di pacchetti di rete per Linux,
  e` disponibile per il download qui:
  http://sourceforge.net/projects/ipfire-wall

* Informazioni e pagina web ufficiale:
  http://www.giacomos.it/iqfire/index.html

- - - - - - - - - - - - - - - - - - - - - -

 . ''  `.
:   :'    :
 `.  ` '
    `- Debian GNU/Linux -- The power of freedom
        http://www.debian.org

bourtemb, 2019-03-12:
> I tried to delete a DeviceProxy object after the ORB had been shutdown. No crash, no error. Could it be that Qt is unloading the Tango library before your library?
> That's quite likely. The [...]

That's a very good question... I don't know how Qt deals with the libraries, especially when several plugins depending on a same library are loaded and unloaded.

> Do you load several similar Tango plugins together?
> Actually maybe the QTango alongside the cumbia one. I could try with one only if you think that makes a difference. Giacomo

I think it's worth a try.

delleceste, 2019-03-12:
OK I Will. so there will be only one Tango plugin loaded by the application.

-- 
Giacomo S.
http://www.giacomos.it

Sincrotrone Trieste S.C.p.A. di interesse nazionale
Strada Statale 14 - km 163,5 in AREA Science Park
34149 Basovizza, Trieste ITALY
-
040 3758073
328 3237959

- - - - - - - - - - - - - - - - - - - - - -

* iqfire-wall, un progetto
  open source che implementa un
  filtro di pacchetti di rete per Linux,
  e` disponibile per il download qui:
  http://sourceforge.net/projects/ipfire-wall

* Informazioni e pagina web ufficiale:
  http://www.giacomos.it/iqfire/index.html

- - - - - - - - - - - - - - - - - - - - - -

 . ''  `.
:   :'    :
 `.  ` '
    `- Debian GNU/Linux -- The power of freedom
        http://www.debian.org

bourtemb, 2019-05-15:
Hi @delleceste ,
Did you try with only one Tango plugin?
Did it crash in that case?

closed 2022-02-24 by t-b