tango-controls/cppTango #541: Possible error in WAttribute::rollback()

udyni, 2019-03-15:
As I was trying to understand how to recover the memorized value of an attribute, I noticed that in WAttribute::rollback() at line 2647 of w_attribute.cpp the rollback of type UShort is instead rolling back the value for data type Double.

Best regards
Michele

bourtemb, 2019-03-15:
Well spotted, @udyni 
Thank you very much for the bug report!

bourtemb, 2019-03-15:
@udyni , did you get annoyed by this bug somehow or did you simply see this bug by reading the code?
In the first case, can you describe your use case and also the behaviour you were observing when the bug occurred?

udyni, 2019-03-15:
I was studying the code to understand where the memorized attributes were
read from the database and set in the attribute write value and just
noticed the error in the code.

bourtemb, 2019-03-18:
After analyzing the source code and doing some simple tests, it looks like this bug can affect the users using WRITE DevUShort attributes (No impact on READ and READ_WRITE attributes it seems).
When an exception is thrown when trying to write the WRITE attribute, the set value is not rolled back correctly in some specific cases (exception in device server write method associated to this attribute, exception thrown because it is not allowed to write this attribute in the current state, exception in read_attr_hardware()).

bourtemb, 2019-03-29:
Thanks again for the bug report @udyni 
This issue should be fixed in #542

closed 2019-03-29 by bourtemb