tango-controls/cppTango #554: Run tests with valgrind (debug memory issues)

mliszcz, 2019-05-24:
While writing a test for #330, I've stumbled on following issue:
* I've had a working testcase with a reproduction (some changes were needed on the device side).
* I've moved some code around (just cleanup/refactoring).
* Now instead of a segfault I'm getting a garbage read from memory.
* The crash is still reproducible if I run DevTest with valgrind.

I'll try to find a workaround but still I'd like to use this opportunity to discuss whether we shall introduce valgrind in CI. Problems I see for now:
* tests may start failing due to timeouts - we would need to extend `sleep`s
* tests may start failing due to undetected memory issues / leaks - we would need to fix them beforehand

Please share your opinions. @bourtemb @Ingvord @t-b

bourtemb, 2019-05-24:
I'm not against introducing valgrind in the CI but this will for sure slow down the tests, which is a bit sad after the improvement you did in another PR to make the compilation faster.
My experience shows that there will be a big work to be done on the tests to make valgrind happy.
Many tests do not free correctly all the memory which has been allocated.
So this would be a big task to fix all the errors reported by valgrind on the tests.
Then, I would say it is just a question of defining priorities but I agree with you it could be very useful.
Users could also refer to the code of the tests to see some good examples of code well written and memory well managed... That would be nice, for sure.

mliszcz, 2019-05-24:
Thanks for your input. How about defining a separate, independent job that will run e..g nightly? This nightly job will not be tied to any particular commit. After a failure, we would need to review commits from previous day.

Another option is to select just a subset of tests that we want to run with valgrind and then use a different device server instance (started with valgrind) in those tests.

Ingvord, 2019-05-24:
I agree that these kind of tests must be run on nightly/weekly basis and not linked with every commit/push

bourtemb, 2019-05-24:
> Thanks for your input. How about defining a separate, independent job that will run e..g nightly? This nightly job will not be tied to any particular commit. 

So the idea would be to have this job running only on specific branches, like tango-9-lts branch for instance or another branch used before to merge to tango-9-lts branch.
> After a failure, we would need to review commits from previous day.
> Another option is to select just a subset of tests that we want to run with valgrind and then use a different device server instance (started with valgrind) in those tests.

If you want to ease the debugging, I think we would need to restart the device servers between each test in order to have one valgrind log per test and not only one valgrind log per device server for the whole sequence of tests.

mliszcz, 2019-05-24:
> If you want to ease the debugging, I think we would need to restart the device servers between each test in order to have one valgrind log per test and not only one valgrind log per device server for the whole sequence of tests.
That is a different topic, I think even more important. But you don't need valgrind logs from CI to *detect* issues. Valgrind can crash the application which seems to work otherwise, due to undefined behavior. You can debug later, locally, and just run a single testcase.

closed 2022-02-24 by t-b