tango-controls/cppTango #563: Remove CodacyBot

t-b, 2019-06-05:
I think the codacybot is too noisy and has little gain as it does not understand C++ good enough. How about we remove it for now?

Ingvord, 2019-06-06:
+1

Moreover we can remove codacy integration at all, as Sonar does much better job and I doubt anyone will ever pay attention to both

mliszcz, 2019-06-06:
> I think the codacybot is too noisy and has little gain as it does not understand C++ good enough.

How about making it less noisy (disable email notifications). Also, could you provide an example of a false positive where the bot incorrectly detected an issue? Maybe it is possible to disable some checks in codacy and leave only critical ones.

> Moreover we can remove codacy integration at all, as Sonar does much better job

The problem with sonar is that it seem to run quite randomly. See e.g. two recent PRs: #528 (no sonar) and #551 (there is sonar check).

Ingvord, 2019-06-06:
> > Moreover we can remove codacy integration at all, as Sonar does much better job

>The problem with sonar is that it seem to run quite randomly. See e.g. two recent PRs: #528 (no sonar) and #551 (there is sonar check).

It is not a problem with Sonar, we just need to invest some more efforts into its configuration. I did what I can due to the time constraints, maybe the robust solution is just around the corner now

t-b, 2019-06-06:
@mliszcz 
See https://app.codacy.com/app/tango-controls/cppTango/pullRequest?prid=3023939

The "Function parameter * should be passed by const reference." is true for C++03 but wrong for these cases in C++11 and later. I also did 

Same link under fixed issues:

Says "Uninitialized variable: lgstr_arr"

```
                            case CORBA::tk_long:
                                Tango::DevVarLongStringArray *lgstr_arr;
                                dd.any.inout() >>= lgstr_arr;
                                o_str << lgstr_arr->lvalue << endl;
                                o_str << lgstr_arr->svalue;
                                break;
```

for the following code which know you might be tempted to just `tango_nullptr` initialize lgstr_arr, but that not a real fix as a real fix should check the `>>=` operation in addition.

mliszcz, 2019-06-06:
Did you mean the `attname` parameter in `get_attr_object_by_name` shall be passed by value?

> Function parameter 'attname' should be passed by const reference.

> Tango::Attr *FwdTestClass::get_attr_object_by_name(std::vector<Tango::Attr *> &att_list, std::string attname)

If yes, this is not true and the bot is right, you are making unnecessary copy.

> Says "Uninitialized variable: lgstr_arr"

No, the bot knows nothing about `>>=` implementation (whether it is ok to pass unintialized object or not). In C++11 you should use `Tango::DevVarLongStringArray *lgstr_arr{};`

bourtemb, 2019-06-06:
Good initiative @t-b!
Good points @mliszcz !
I use to think as @t-b for the uninitialized variable report, but indeed, you're right, the bot does not know whether it is ok to pass uninitialized object to >>= operator or not.

I think codacy is indeed redundant with sonar and I agree to remove it completely.
But as suggested by @ingvord, someone has to look at sonarcloud configuration to ensure that it is run for every PR.

Any volunteer?

t-b, 2019-06-06:
@bourtemb Okay, I'll give that a shot.

bourtemb, 2019-06-06:
> @bourtemb Okay, I'll give that a shot.

Thanks! 
I don't manage to add you in the list of members/admin for sonarcloud project.
When I try to add you from the sonarcloud interface, I see the name of other users containing the string 't-b' in their github account names, but I don't see you...

bourtemb, 2019-06-06:
@t-b you need to have signed up on the sonarcloud service so I could see you and add you as admin, so you could be able to change/see all the settings.

t-b, 2019-06-06:
@bourtemb Done. I just registered.

bourtemb, 2019-06-06:
@t-b You should have the admin privileges now.

t-b, 2019-07-25:
@t-b Can you add me for codacy as well? I can't remove the project right now.

t-b, 2019-10-22:
Removed already.

closed 2019-10-22 by t-b