tango-controls/cppTango #566: Use const parameters for operator<<(ostream &, ThisObjectShouldBeConst)

bourtemb, 2019-06-12:
Hi,

We should add the const qualifier for the 2nd parameter of the ostream operator<<(ostream &, ...) functions.
For instance:
`ostream &operator<<(ostream &o_str, AttributeInfoEx &p)`
should be:
`ostream &operator<<(ostream &o_str, const AttributeInfoEx &p)`

There are many other cases to be fixed.
Thanks @chedburgh for the bug report.

bourtemb, 2019-06-12:
Adding the const qualifier will break the binary compatibility but I think there are some other pending changes which will also require a binary compatibility break so we can include this change in the same version as these other pending changes.

mliszcz, 2019-07-22:
impacted files:
```
rg 'operator<<\((std::)?ostream' cppapi/ -l
```

* [ ] cppapi/server/attribute.h
* [ ] cppapi/server/attrmanip.cpp
* [ ] cppapi/server/attrdesc.h
* [ ] cppapi/server/seqvec.h
* [ ] cppapi/server/classattribute.cpp
* [ ] cppapi/server/templ_inst.cpp
* [ ] cppapi/server/classattribute.h
* [ ] cppapi/server/attrmanip.h
* [ ] cppapi/server/seqvec.cpp
* [ ] cppapi/server/blackbox.h
* [ ] cppapi/server/blackbox.cpp
* [ ] cppapi/client/devapi.h
* [ ] cppapi/server/attribute.cpp
* [ ] cppapi/client/api_util.cpp
* [ ] cppapi/client/devapi_attr.cpp
* [ ] cppapi/client/DevicePipe.h
* [ ] cppapi/client/DeviceData.h
* [ ] cppapi/client/devapi_utils.cpp
* [ ] cppapi/client/devapi_pipe.tpp
* [ ] cppapi/client/devapi_pipe.cpp
* [ ] cppapi/client/devapi_data.cpp
* [ ] cppapi/client/DeviceAttribute.h
* [ ] cppapi/client/devapi_datahist.cpp

mliszcz, 2019-07-22:
There will be a side effect of this change. We have functions which are not const, but should be (they are "queries"), like `da.has_failed()` in code below. It should not change `da`'s state:

```cpp
ostream &operator<<(ostream &o_str, const DeviceAttribute &da)
{
	if (da.has_failed() == true)
```

PR #574 (2020-09-07): Fix 566 const reference parameters
Following chages are included:
* pass object to print by `const&` in all `operator<<(std::ostream&, ...)`
* mark some query methods as `const`
* mark `DeviceAttribute::ext::ext_state` as `mutable` (it is updated from e.g. `Attribute::is_empty`

Fixes #566
commit 40c05aef (2020-09-07): Merge pull request #574 from mliszcz/fix-566-const-reference-parameters

We now correctly pass const references when we print objects with operator<<(ostream &, ThisObjectShouldBeConst).

Fixes #566
closed 2020-09-07 by mliszcz