tango-controls/cppTango #583: Move data members and methods which were added in ext classes into their corresponding Tango class

bourtemb, 2019-07-31:
Since the next cppTango version will not be binary compatible with 9.3.3, we should take this opportunity to move all the features (data members/ methods) which have been implemented in some ext classes into their corresponding Tango class.

An example is the MultiAttributeExt class which should be emptied and its features moved to MultiAttribute class.

mliszcz, 2019-12-17:
Just copying my comment from slack discussion with Reynald:

> Regarding 583, what is the reason for moving contents of Ext classes into their parent classes? E.g. currently there is an omni_mutex in AttributeExt. If we move the mutex to Attribute, and at some point in the future we decide that the mutex must be removed (e.g. due to some refactoring), or changed to std::mutex, we will break the ABI again.  Keeping private data members in a hidden class solves this issue and is a well-known idiom in C++

Here are more resources on Pimpl:
https://www.bfilipek.com/2018/01/pimpl.html
https://en.cppreference.com/w/cpp/language/pimpl

t-b, 2019-12-17:
+1 for pimpl.

closed 2022-02-24 by t-b