tango-controls/cppTango #586: Use automatic code formatting

t-b, 2019-08-01:
Inspired from https://github.com/tango-controls/cppTango/pull/581#issuecomment-516828044

Situation:
 The code base is currently manually formatted and indented. So it sometimes looks like

[image]

This is of course not nice as it forces the developer to manually indent using mixed indentation. And it is also a waste of developers time.

Proposal:
- [ ] Add clang-format file from https://github.com/AllenInstitute/ZeroMQ-XOP/blob/master/src/.clang-format
- [ ] Add a cmake target which calls clang-format
- [ ] Start discussion about how formatting should be done
- [ ] Make **1** commit changing all the C++ code to use the new formatting
- [ ] Document how users can ignore this commit when blaming using the new git blame feature available from git 2.23 (not yet released), see also https://github.com/gitster/git/commit/ae3f36dea16e51041c56ba9ed6b38380c8421816.
- [ ] Add a travis job which ensure that the PR matches our code formatting. Passing this check is *mandatory* for all PRs.
- [ ] Finalize formatting discussion

Ingvord, 2019-08-01:
+1

My 50 cents: auto-formatting procedure must be as simple as `Ctrl+Alt+L` in CLion

bourtemb, 2019-08-02:
> 
> 
> +1
> 
> My 50 cents: auto-formatting procedure must be as simple as `Ctrl+Alt+L` in CLion

It looks like CLion already supports clang-format:
https://www.jetbrains.com/help/clion/clangformat-as-alternative-formatter.html

closed 2022-02-24 by t-b