tango-controls/cppTango #588: Add support for DevState memorized attributes

bourtemb, 2019-08-12:
As noticed by @mliszcz in https://github.com/tango-controls/rfc/pull/32, it seems like DevState memorized attributes are not allowed.
I don't think there is any good reason for not supporting them.

Attr:set_memorized() method seems to forbid memorized attributes of type STATE:
https://github.com/tango-controls/cppTango/blob/f3efe02db1ef1bfd5347901f0e7b5e20aba478da/cppapi/server/attrdesc.cpp#L665-L675

I think this is a mistake in the current implementation. Attributes of type DevState (other than attribute named "State") were not there at the beginning when attributes were created in Tango. It looks like the test in Attr::set_memorized() has not been updated when the support for DevState user attributes was added.
In theory, there is no reason not to support STATE memorized attributes and we can even see that some other methods are ready for that, like Device_3Impl::write_attributes_in_db for instance:
https://github.com/tango-controls/cppTango/blob/tango-9-lts/cppapi/server/device_3.cpp#L2296-L2382

We should add a test for DevState memorized attributes.

lorenzopivetta, 2019-11-19:
So, my understanding is support memorized DevState attributes **other than** State, which is appropriate as in my view the State attribute value is synthesized by the device based on inputs and logic (and not restored from saved value). E.g. State will make an exception concerning DevState memorized attr.  Is this correct?

bourtemb, 2019-11-19:
> So, my understanding is support memorized DevState attributes **other than** State

@lorenzopivetta, this is indeed for DevState attribute other than the Attribute named "State".
The "State" attribute is considered as READ only in any case so it would not make sense to memorize a READ Only attribute.

lorenzopivetta, 2019-11-19:
Right. Just wanted to stress it, and clarify that the causality is "State is RO because by design as it is supposed to be managed by the device" and not "cannot memorize state because it is RO", which is true, indeed, but as a consequence.

closed 2022-02-24 by t-b