tango-controls/cppTango #594: Prepare release 9.3.4

bourtemb, 2019-10-21:
Task list:
- [x] Read/review steps listed in the ["Releasing a new version" documentation](https://tango-controls.readthedocs.io/en/latest/development/contributing/cpptango-contribution.html#releasing-a-new-version) for making a release
- [ ] Bump the version (if not already done) on 9.3-backports branch
- [x] Merge the wished Pull Requests into 9.3-backports (Make sure Travis and appveyor are OK on the PR before the merge)
- [ ] Make sure the documentation is updated (readthedocs, doxygen)
- [x] Make sure Travis and appveyor are OK on 9.3-backports branch
- [ ] Edit the CHANGELOG.md file
- [ ] Edit the RELEASE_NOTES file
- [ ] Create a release tag on GitHub, from 9.3-backports branch. Flag it as a pre-release.
- [ ] Advertise the pre-release on the mailing list to ask for feedback and testers
- [ ] Wait for at least 2 weeks before to do the next steps to give some time to get feedback from the first testers
- [ ] If no major bug was reported, the GitHub release can be flagged as "Latest release",
else the bugs should be fixed first and a new pre-release should be created.
- [ ] Add Windows installers as assets to the GitHub release
- [ ] Generate Doxygen documentation for this new release and update cppTango-docs with the new generated documentation
- [ ] Update cppTango version in TangoSourceDistribution repository
- [ ] Release a new version of the TangoSourceDistribution
- [ ] Upgrade [tango-cs docker](https://github.com/tango-controls/tango-cs-docker) and eventually [tango-db docker](https://github.com/tango-controls/docker-mysql) (if a new version of TangoDatabase has been released).
- [ ] Advertise the release as the new stable release on the mailing list and on the tango-controls website (News)
- [ ] Close this issue

bourtemb, 2019-10-21:
In this specific case, since this release is there to fix a problem when building the source distribution on Debian. I think we should work with @picca and provide asap a TangoSourceDistribution 9.3.4-rc1 he could test to ensure the problem is indeed fixed. We should not wait for 2 weeks to prepare the 9.3.4-rc1 source distribution release candidate of course.  ;-)

bourtemb, 2019-10-21:
@mliszcz, @piogor 
For tango-doc (readthedocs), since the documentation didn't change between cppTango 9.3.3 and 9.3.4.
What will we need to do?
* Create a 9.3.4 branch from 9.3.3 branch?
* Update source/conf.py release like in this commit: https://github.com/tango-controls/tango-doc/commit/64e32793197ab59646a115a37391c05338ab6c57 ?
* Update links to the 9.3.4 source distribution
* ... ?

t-b, 2019-10-21:
@bourtemb Yes no need to wait. We can always do another rc if something comes up.

bourtemb, 2019-10-21:
I just created this PR: https://github.com/tango-controls/cppTango/pull/596 to fix the Debian 7 Travis tests.

After merging this PR #596, it might be a good idea to setup Travis tests for Debian Buster on this 9.3-backports branch since we are trying to solve an issue related to Debian Buster.

Maybe doing a git cherry-pick of these 2 commits would be enough?:

https://github.com/tango-controls/cppTango/pull/562/commits/4e259462df47141df928ab99993bfd6157ebe5fb
https://github.com/tango-controls/cppTango/pull/562/commits/86174eea96f54b7b8d0962029997154a0913e5e8

@t-b, can you do that?

t-b, 2019-10-21:
@bourtemb Yeep, will do.

mliszcz, 2019-10-22:
@bourtemb, regarding tango-doc, what you listed is enough IMO. Note that master branch is 12 commits ahead of 9.3.3. Almost all of these commits still apply to 9.3.4 so we can branch-out from master.

The commit https://github.com/tango-controls/tango-doc/commit/c02e173d2bac5a6960bde6dbff049c27281d7dd2 looks suspicious to me, as it is quite a big merge commit for some layout redesign, but 9.3.3 and master have the same layout. Maybe @piogor can comment.

bourtemb, 2019-10-22:
> After merging this PR #596, it might be a good idea to setup Travis tests for Debian Buster on this 9.3-backports branch since we are trying to solve an issue related to Debian Buster.
> 
> Maybe doing a git cherry-pick of these 2 commits would be enough?:
> 
> [4e25946](https://github.com/tango-controls/cppTango/commit/4e259462df47141df928ab99993bfd6157ebe5fb)
> [86174ee](https://github.com/tango-controls/cppTango/commit/86174eea96f54b7b8d0962029997154a0913e5e8)
> 
> @t-b, can you do that?

These commits are using some CMake options which are not available in 9.3-backports branch, so we will have to adapt them.

t-b, 2019-10-22:
@bourtemb I've created #597 and also adapted some of the code to not reference not-existing cmake flags.

t-b, 2019-10-22:
@bourtemb Appveyor seems not to be running for the 9.3-backports branch. I can have a look if we you me on appveyor.

bourtemb, 2019-10-22:
> @bourtemb Appveyor seems not to be running for the 9.3-backports branch. I can have a look if you add me on appveyor.

You just need to play with the appveyor.yml file (https://github.com/tango-controls/cppTango/blob/9.3-backports/appveyor.yml)

At the beginning of the file, we simply need to add 9.3-backports branch to the list of branches.

t-b, 2019-10-22:
@bourtemb Done in #600.

t-b, 2019-10-23:
@bourtemb Regarding "Add Windows installers as assets to the GitHub release": Is that currently done with a script (where is that?) or if not, is it worth having one?

t-b, 2019-10-23:
Appveyor does also support that out of the box. See https://www.appveyor.com/docs/deployment/github/.

bourtemb, 2019-10-23:
> @bourtemb Regarding "Add Windows installers as assets to the GitHub release": Is that currently done with a script (where is that?) or if not, is it worth having one?

I think it was done manually the last time. Automating this process would be much better of course.

> Appveyor does also support that out of the box. See https://www.appveyor.com/docs/deployment/github/.

Good idea. I think we should try that.

mliszcz, 2019-10-28:
Hi @t-b and @bourtemb 

Regarding: "Update the upgrade scripts in the TangoDatabase repository (tango-controls/TangoDatabase#19)"

The same needs to be done in TangoSourceDistribution. All relevant files were touched https://github.com/tango-controls/TangoSourceDistribution/commit/46722ce99c12527a088b23c01e8d74e94f572880

bourtemb, 2019-10-28:
> 
> 
> Hi @t-b and @bourtemb
> 
> Regarding: "Update the upgrade scripts in the TangoDatabase repository ([tango-controls/TangoDatabase#19](https://github.com/tango-controls/TangoDatabase/pull/19))"
> 
> The same needs to be done in TangoSourceDistribution. All relevant files were touched [tango-controls/TangoSourceDistribution@46722ce](https://github.com/tango-controls/TangoSourceDistribution/commit/46722ce99c12527a088b23c01e8d74e94f572880)

Good point @mliszcz ! This has to be done too indeed.

t-b, 2019-10-28:
@bourtemb @mliszcz Done as part of https://github.com/tango-controls/TangoSourceDistribution/pull/53.

bourtemb, 2019-10-29:
I think we can already tag a 9.3.4-rc1 in cppTango.

What about the appveyor artifacts deployments?
Do we do try it in that branch?

> I think it was done manually the last time. Automating this process would be much better of course.
>
>
> Good idea. I think we should try that.

t-b, 2019-10-29:
@bourtemb Yes I would try the appveyor deployment for that new tag. I just need to find out if I have to create the tag before the appyeyor config or afterwards.

Here is what I will do

```
firma@thomas:~/devel/cppTango (9.3-backports)
$git tag 9.3.4-rc1
firma@thomas:~/devel/cppTango (9.3-backports)
$git push  --tags --dry-run
To github.com:tango-controls/cppTango.git
 * [new tag]           9.3.4-rc1 -> 9.3.4-rc1
```

for creating the tag. Without the `--dry-run` of course ;)

bourtemb, 2019-10-29:
> @bourtemb Yes I would try the appveyor deployment for that new tag. I just need to find out if I have to create the tag before the appyeyor config or afterwards.

I think you can already create and push an 9.3.4-rc1 tag.
You can work on the appveyor config afterwards. We will create an new tag when we will do the stable release.

For the appveyor config. I would advise to test it on your own fork first so you can create as many tags as you want there during the tests.

t-b, 2019-11-04:
I've just created https://github.com/tango-controls/TangoSourceDistribution/releases/tag/9.3.4-rc1.

bourtemb, 2019-11-04:
@picca could you please give this new distribution release candidate a try?

picca, 2019-11-04:
work in progress

https://salsa.debian.org/science-team/tango/pipelines/83924

cheers

Fred

picca, 2019-11-04:
Hello, I have this warning  during the build.

StarterUtil.cpp:233:34: warning: ‘%02d’ directive writing between 2 and 11 bytes into a region of size between 0 and 9 [-Wformat-overflow=]
  233 |  sprintf (str, "%02d/%02d/%02d   %02d:%02d:%02d",
      |                                  ^~~~
StarterUtil.cpp:233:10: note: ‘sprintf’ output between 20 and 74 bytes into a destination of size 20
  233 |  sprintf (str, "%02d/%02d/%02d   %02d:%02d:%02d",
      |  ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  234 |         st->tm_mday, st->tm_mon+1, st->tm_year,
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  235 |         st->tm_hour, st->tm_min, st->tm_sec ) ;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

picca, 2019-11-04:
```
g++ -DHAVE_CONFIG_H -I. -I../..  -I/usr/include  -isystem /usr/include/mit-krb5 -I/usr/include/pgm-5.2 -I../../lib/cpp/log4tango/include -I../../lib/cpp/log4tango/include -I../../lib/cpp/server -I../../lib/cpp/client -I../../lib/cpp/server -I/usr/include/mysql -I/usr/include -Wl,-z,relro -I/usr/include  -Wall -D_FORTIFY_SOURCE=2 -O1 -fPIE -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security   -D_REENTRANT -DOMNI_UNLOADABLE_STUBS -c -o main.o main.cpp
g++ -DHAVE_CONFIG_H -I. -I../..  -I/usr/include  -isystem /usr/include/mit-krb5 -I/usr/include/pgm-5.2 -I../../lib/cpp/log4tango/include -I../../lib/cpp/log4tango/include -I../../lib/cpp/server -I../../lib/cpp/client -I../../lib/cpp/server -I/usr/include/mysql -I/usr/include -Wl,-z,relro -I/usr/include  -Wall -D_FORTIFY_SOURCE=2 -O1 -fPIE -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security   -D_REENTRANT -DOMNI_UNLOADABLE_STUBS -c -o update_starter.o update_starter.cpp
main.cpp: In function ‘int main(int, char**)’:
main.cpp:220:9: warning: catching polymorphic type ‘class std::bad_alloc’ by value [-Wcatch-value=]
  220 |  catch (bad_alloc)
      |         ^~~~~~~~~
main.cpp: At global scope:
main.cpp:2:20: warning: ‘RcsId’ defined but not used [-Wunused-variable]
    2 | static const char *RcsId = "$Id$";
      |                    ^~~~~
g++ -DHAVE_CONFIG_H -I. -I../..  -I/usr/include  -isystem /usr/include/mit-krb5 -I/usr/include/pgm-5.2 -I../../lib/cpp/log4tango/include -I../../lib/cpp/log4tango/include -I../../lib/cpp/server -I../../lib/cpp/client -I../../lib/cpp/server -I/usr/include/mysql -I/usr/include -Wl,-z,relro -I/usr/include  -Wall -D_FORTIFY_SOURCE=2 -O1 -fPIE -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security   -D_REENTRANT -DOMNI_UNLOADABLE_STUBS -c -o DataBaseUtils.o DataBaseUtils.cpp
DataBaseUtils.cpp:1:20: warning: ‘RcsId’ defined but not used [-Wunused-variable]
    1 | static const char *RcsId = "$Header$";
      |                    ^~~~~
DataBase.cpp: In member function ‘virtual void DataBase_ns::DataBase::read_Timing_info(Tango::Attribute&)’:
DataBase.cpp:724:55: warning: ‘%s’ directive writing up to 255 bytes into a region of size 221 [-Wformat-overflow=]
  724 |  sprintf(info_str,"TANGO Database Timing info on host %s",hostname);
      |                                                       ^~  ~~~~~~~~
In file included from /usr/include/stdio.h:867,
                 from /usr/include/c++/9/cstdio:42,
                 from /usr/include/c++/9/ext/string_conversions.h:43,
                 from /usr/include/c++/9/bits/basic_string.h:6493,
                 from /usr/include/c++/9/string:55,
                 from ../../lib/cpp/server/tango.h:39,
                 from DataBase.cpp:48:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:36:34: note: ‘__builtin___sprintf_chk’ output between 36 and 291 bytes into a destination of size 256
   36 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   37 |       __bos (__s), __fmt, __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

a few more.

picca, 2019-11-04:
Hello can you give the command line in order to generate the documentation from the rst files ?

thanks

Frederic

picca, 2019-11-04:
Ici vous avez les erreurs lintian.

Il me semble que le plus problematique c'est le numero de version so.
CEq ue vous avez mis dans le fichier configure.ac genere une librairie

libtango.so.8.4.3

https://salsa.debian.org/science-team/tango/-/jobs/399174

Frederic

picca, 2019-11-04:
The previous one 

libtango.so.9.2.5

t-b, 2019-11-04:
@picca There seems to be a problem with the SONAME?

Generating the documentation could be done with something like https://github.com/tango-controls/tango-doc/blob/7f4829d8e73cadb286570393c8e681b49bc857c5/.travis.yml#L11.

Regarding the warnings. Yes this is expected. You seem to use a different set of compile warnings. Can you post these?

picca, 2019-11-04:
> @picca<https://github.com/picca> There seems to be a problem with the SONAME?

Yes, I think that the number is wrong

you did this

-VERSION_INFO=12:3:3
+VERSION_INFO=12:3:4

Here the doc for this number upgrade

https://www.gnu.org/software/libtool/manual/html_node/Updating-version-info.html

> Generating the documentation could be done with something like https://github.com/tango-controls/tango-doc/blob/7f4829d8e73cadb286570393c8e681b49bc857c5/.travis.yml#L11.

ok, I will build the doc with sphinx-build

> Regarding the warnings. Yes this is expected. You seem to use a different set of compile warnings. Can you post these?

here all the build log with the flags :)

https://salsa.debian.org/science-team/tango/-/jobs/399170

Cheers

t-b, 2019-11-04:
I'll have a look regarding the SONAME.

I've seen the logs arlready. But how do you set the additional build flags? I've found https://wiki.debian.org/Hardening which says that `dh` does that magic? Or is it `dpkg-buildpackage`?

picca, 2019-11-04:
> I've seen the logs arlready. But how do you set the additional build flags? I've found https://wiki.debian.org/Hardening which says that dh does that magic? Or is it dpkg-buildpackage?

Yes, I use dh in order to do the job.

I would do 

export DEB_BUILD_MAINT_OPTIONS = hardening=+all

in order to activate all the hardening flags.

picca, 2019-11-04:
I use this

https://wiki.debian.org/HardeningWalkthrough#Enabling_dpkg-buildflags_in_your_debian.2Frules_files

picca, 2019-11-04:
If you look here, you will see that a few files are not compiled with the right hardening flags.

https://salsa.debian.org/science-team/tango/-/jobs/399228

picca, 2019-11-04:
Looking at the cppflags, I have this during the configuration

=== configuring in lib/cpp/log4tango (/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/lib/cpp/log4tango)
configure: running /bin/bash ./configure --disable-option-checking '--prefix=/usr'  '--build=x86_64-linux-gnu' '--includedir=${prefix}/include' '--mandir=${prefix}/share/man' '--infodir=${prefix}/share/info' '--sysconfdir=/etc' '--localstatedir=/var' '--disable-silent-rules' '--libdir=${prefix}/lib/x86_64-linux-gnu' '--runstatedir=/run' '--disable-maintainer-mode' '--disable-dependency-tracking' '--with-omni=/usr' '--disable-static' '--disable-dbcreate' '--disable-java' '--with-doxygen' '--disable-jpegmmx' 'build_alias=x86_64-linux-gnu' 'CXXFLAGS=-g -O2 -fdebug-prefix-map=/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1=. -fstack-protector-strong -Wformat -Werror=format-security' 'LDFLAGS=-Wl,-z,relro -Wl,-z,now' 'CPPFLAGS=-Wdate-time -D_FORTIFY_SOURCE=2' 'CFLAGS=-g -O2 -fdebug-prefix-map=/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1=. -fstack-protector-strong -Wformat -Werror=format-security' --cache-file=/dev/null --srcdir=.

you can see the right  'CPPFLAGS=-Wdate-time -D_FORTIFY_SOURCE=2' .

this flag is used by log4tango, but not during the build of tango.
During the reconfiguration, it seems that this 

src/Makefile.am:3: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/Makefile.am: installing 'config/depcomp'
tests/Makefile.am:11: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')

So maybe this is why the CPPGLAGS is not used by tango...

t-b, 2019-11-04:
@picca This alone does not solve the problem. But my other fixes also don't yet work.

t-b, 2019-11-04:
For now I you can work around that issue by passing these flags in CXXFLAGS.

picca, 2019-11-04:
I will try to patch configure.ac tomorrow and see if it works :)

t-b, 2019-11-04:
Current state is at https://github.com/tango-controls/TangoSourceDistribution/pull/55.

The really odd thing is that log4tango works when you do

```
$./configure CPPFLAGS=-D_FORTIFY_SOURCE=2
$find . -iname "config.log" | xargs grep "^CPPFLAGS="
./lib/cpp/log4tango/config.log:CPPFLAGS='-D_FORTIFY_SOURCE=2'
./config.log:CPPFLAGS=' -I/usr/local/include '
firma@thomas:~/devel/TangoSourceDistribution/build/distr (debian-build-fixes)
```

t-b, 2019-11-05:
My guess is that something nasty is done in one of the files in https://github.com/tango-controls/TangoSourceDistribution/tree/master/assets/m4, but I do need fresh eyes for that...

picca, 2019-11-05:
I regenerated the autoconf files and ebd up with this.

libtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:315: warning: macro 'AM_PATH_MYSQLCLIENT' not found in library
configure.ac:368: warning: macro 'AM_PATH_MARIADBCLIENT' not found in library
configure.ac:82: error: possibly undefined macro: gcc_AC_HAVE_GCC_VERSION
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:113: error: possibly undefined macro: java_AC_HAVE_JAVA_VERSION
configure.ac:284: error: possibly undefined macro: AC_CXX_HAVE_CLASS_STRSTREAM
configure.ac:315: error: possibly undefined macro: AM_PATH_MYSQLCLIENT
configure.ac:330: error: possibly undefined macro: mysql_AC_HAVE_MYSQL_VERSION
configure.ac:368: error: possibly undefined macro: AM_PATH_MARIADBCLIENT
configure.ac:383: error: possibly undefined macro: mariadb_AC_HAVE_MARIADB_VERSION
autoreconf: /usr/bin/autoconf failed with exit status: 1

So I guess that the culprite is one of these macro.

nevertheless, maybe this could be someting in the Makefile.am

picca, 2019-11-05:
looking at the config.log, I found this in the zlib test

```
if test -n "${ZLIB_HOME}"
then
        ZLIB_OLD_LDFLAGS=$LDFLAGS
        ZLIB_OLD_LIBS=$LIBS
        ZLIB_OLD_CPPFLAGS=$LDFLAGS

        LDFLAGS="$LDFLAGS -L${ZLIB_HOME}/lib"
        CPPFLAGS="$CPPFLAGS -I${ZLIB_HOME}/include"
        CPPFLAGS="$CPPFLAGS -I${ZLIB_HOME}/include"

        ac_ext=c
ac_cpp='$CPP $CPPFLAGS'
ac_compile='$CC -c $CFLAGS $CPPFLAGS conftest.$ac_ext >&5'
ac_link='$CC -o conftest$ac_exeext $CFLAGS $CPPFLAGS $LDFLAGS conftest.$ac_ext $LIBS >&5'
ac_compiler_gnu=$ac_cv_c_compiler_gnu
```

and at the end of the test

```
        LDFLAGS="$ZLIB_OLD_LDFLAGS"
        CPPFLAGS="$ZLIB_OLD_CPPFLAGS"
        LIBS="$ZLIB_OLD_LIBS"
```

so for now my guess is that there is a problem in the ZLIB test.

        ZLIB_OLD_CPPFLAGS=$LDFLAGS

should be 

        ZLIB_OLD_CPPFLAGS=$CPPFLAGS

picca, 2019-11-05:
What about using pkg-config as much as possible in stead these very very old checks.

picca@sixs7:~/Debian/tango/tango/m4$ pkg-config --libs zlib
-lz

picca, 2019-11-05:
bingo

https://salsa.debian.org/science-team/tango/-/jobs/400247

I replaces

 # check for the presence of zlib
-CHECK_ZLIB
+PKG_CHECK_MODULES([ZLIB], [zlib >= 1])
 
and it works.

Cheers

t-b, 2019-11-05:
@picca Good find! I've updated my branch.

picca, 2019-11-05:
Hello, I build the documentation via the Makefile in the doc/src/

this produce the documentation under build/html

Which is not the location were you put the pre-build documentation.

I think that it would be better to build the documentation at the right place :)

picca, 2019-11-05:
Here a bunch of warning due to the doc.

/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/administration/services/hdbpp/hdb++-design-guidelines.rst:9: WARNING: term not in glossary: hdb++
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/administration/services/hdbpp/index.rst:8: WARNING: term not in glossary: hdb++
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/development/contributing/documentation-guide.rst:81: WARNING: unknown document: tutorials/documentation-workflow-tutorial
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/getting-started/development/java/jtango-server.rst:90: WARNING: unknown document: ../../../tools-and-extensions/pogo/index
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/getting-started/first-steps.rst:14: WARNING: unknown document: intsallation/index
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/installation/tango-on-windows.rst:132: WARNING: term not in glossary: device servers
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/installation/tango-on-windows.rst:140: WARNING: term not in glossary: device servers
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/reference/glossary.rst:53: WARNING: term not in glossary: attribute quality
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tools-and-extensions/built-in/astor/config.rst:23: WARNING: unknown option: -DNO_PREF=true
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tools-and-extensions/built-in/atk/atk.rst:513: WARNING: Could not lex literal_block as "java". Highlighting skipped.
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tools-and-extensions/built-in/atk/atk.rst:573: WARNING: Could not lex literal_block as "java". Highlighting skipped.
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tools-and-extensions/built-in/atk/atk.rst:599: WARNING: Could not lex literal_block as "java". Highlighting skipped.
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tools-and-extensions/built-in/atk/atk.rst:961: WARNING: Could not lex literal_block as "java". Highlighting skipped.
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tools-and-extensions/built-in/atk/atk.rst:1046: WARNING: Could not lex literal_block as "java". Highlighting skipped.
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tools-and-extensions/built-in/atk/atk.rst:1477: WARNING: Could not lex literal_block as "java". Highlighting skipped.
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tools-and-extensions/built-in/atk/atk.rst:1486: WARNING: Could not lex literal_block as "java". Highlighting skipped.
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tools-and-extensions/built-in/atk/atk.rst:1495: WARNING: Could not lex literal_block as "java". Highlighting skipped.
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tools-and-extensions/built-in/jdraw/jdraw.rst:281: WARNING: Could not lex literal_block as "java". Highlighting skipped.
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tutorials-and-howtos/how-tos/how-to-create-inheritance-link.rst:64: WARNING: Could not lex literal_block as "cpp". Highlighting skipped.
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tutorials-and-howtos/how-tos/how-to-start-device-server.rst:14: WARNING: undefined label: astro_manual (if the link has no caption the label must precede a section header)
/builds/science-team/tango/debian/output/tango-9.3.4~rc1+dfsg1/doc/src/source/tutorials-and-howtos/how-tos/how-use-vectors-set-attributes.rst:42: WARNING: Could not lex literal_block as "python". Highlighting skipped.

t-b, 2019-11-05:
@picca Thx. https://github.com/tango-controls/TangoSourceDistribution/pull/55/commits now has the fix for the wrong SONAME as well.

picca, 2019-11-05:
while working on the packaging I got these error messages  via lintian

```
warning: the authors of lintian do not recommend running it with root privileges!
P: tango source: source-contains-prebuilt-flash-object doc/html/_static/astor_new_server_wizard.swf
E: tango source: source-is-missing doc/html/_static/astor_new_server_wizard.swf
P: tango source: insane-line-length-in-source-file doc/html/_static/jquery.js line length is 32033 characters (>512)
P: tango source: source-contains-prebuilt-javascript-object doc/html/_static/jquery.js line length is 32033 characters (>512)
E: tango source: source-is-missing doc/html/_static/jquery.js line length is 32033 characters (>512)
P: tango source: insane-line-length-in-source-file doc/html/_static/underscore.js line length is 519 characters (>512)
P: tango source: source-contains-prebuilt-javascript-object doc/html/_static/underscore.js line length is 519 characters (>512)
E: tango source: source-is-missing doc/html/_static/underscore.js line length is 519 characters (>512)
P: tango source: source-contains-prebuilt-javascript-object doc/html/_static/js/modernizr.min.js
E: tango source: source-is-missing doc/html/_static/js/modernizr.min.js
P: tango source: source-contains-prebuilt-javascript-object doc/html/_static/js/theme.js line length is 505 characters (>256)
E: tango source: source-is-missing doc/html/_static/js/theme.js line length is 505 characters (>256)
P: tango source: source-contains-prebuilt-flash-object doc/src/source/_static/astor_new_server_wizard.swf
E: tango source: source-is-missing doc/src/source/_static/astor_new_server_wizard.swf
P: tango source: package-does-not-install-examples doc/src/source/development/client-api/atk/examples/
P: tango source: source-contains-prebuilt-javascript-object doc/src/source/_theme/sphinx_rtd_theme/static/js/modernizr.min.js
E: tango source: source-is-missing doc/src/source/_theme/sphinx_rtd_theme/static/js/modernizr.min.js
P: tango source: source-contains-prebuilt-javascript-object doc/src/source/_theme/sphinx_rtd_theme/static/js/theme.js line length is 505 characters (>256)
E: tango source: source-is-missing doc/src/source/_theme/sphinx_rtd_theme/static/js/theme.js line length is 505 characters (>256)
W: tango source: sphinxdoc-but-no-sphinxdoc-depends
P: tango source: rules-requires-root-missing
I: tango source: wildcard-matches-nothing-in-dep5-copyright lib/cpp/client/zmq.hpp (paragraph at line 15)
W: tango source: global-files-wildcard-not-first-paragraph-in-dep5-copyright (paragraph at line 24)
I: tango source: unused-file-paragraph-in-dep5-copyright paragraph at line 11
I: tango source: unused-file-paragraph-in-dep5-copyright paragraph at line 15
I: tango source: unused-file-paragraph-in-dep5-copyright paragraph at line 19
I: tango source: unused-file-paragraph-in-dep5-copyright paragraph at line 7
I: tango source: debian-rules-parses-dpkg-parsechangelog (line 7)
I: tango source: debian-rules-parses-dpkg-parsechangelog (line 8)
I: tango source: out-of-date-standards-version 4.3.0 (released 2018-12-23) (current is 4.4.1)
W: tango source: spelling-error-in-patch-description debian/patches/0003-use-pkg-config-for-zlib.patch trailling trailing
I: tango source: testsuite-autopkgtest-missing
I: tango-starter: spelling-error-in-binary usr/lib/tango/Starter catched caught
I: tango-starter: spelling-error-in-binary usr/lib/tango/Starter controled controlled
I: tango-starter: spelling-error-in-binary usr/lib/tango/Starter registred registered
I: tango-starter: spelling-error-in-binary usr/lib/tango/Starter automaticly automatically
W: tango-starter: skip-systemd-native-flag-missing-pre-depends prerm:5
W: tango-starter: skip-systemd-native-flag-missing-pre-depends postinst:12
P: tango-starter: missing-systemd-service-for-init.d-script tango-starter
I: libtango-tools: extended-description-is-probably-too-short
W: liblog4tango5v5-dbgsym: debug-file-with-no-debug-symbols usr/lib/debug/.build-id/a4/da39b09f617e1e36f92d5467da684dd8459fe9.debug
I: tango-accesscontrol: hardening-no-fortify-functions usr/lib/tango/TangoAccessControl
W: tango-accesscontrol: skip-systemd-native-flag-missing-pre-depends postinst:12
W: tango-accesscontrol: skip-systemd-native-flag-missing-pre-depends prerm:5
P: tango-accesscontrol: missing-systemd-service-for-init.d-script tango-accesscontrol
I: tango-db: spelling-error-in-binary usr/lib/tango/DataBaseds Propery Property
I: tango-db: spelling-error-in-binary usr/lib/tango/DataBaseds reseting resetting
W: tango-db: no-debconf-templates
W: tango-db: skip-systemd-native-flag-missing-pre-depends postinst:26
W: tango-db: skip-systemd-native-flag-missing-pre-depends prerm:11
P: tango-db: missing-systemd-service-for-init.d-script tango-db
W: liblog4tango5v5: shared-lib-without-dependency-information usr/lib/x86_64-linux-gnu/liblog4tango.so.5.0.2
I: liblog4tango5v5: no-symbols-control-file usr/lib/x86_64-linux-gnu/liblog4tango.so.5.0.2
I: liblog4tango-doc: spelling-error-in-manpage usr/share/man/man3/log4tango::Appender.3.gz occured occurred
I: liblog4tango-doc: spelling-error-in-manpage usr/share/man/man3/log4tango::OstreamAppender.3.gz occured occurred
I: liblog4tango-doc: spelling-error-in-manpage usr/share/man/man3/log4tango::StringUtil.3.gz contructed constructed
I: liblog4tango-doc: spelling-error-in-manpage usr/share/man/man3/log4tango::StringUtil.3.gz contructed constructed
W: liblog4tango-doc: manpage-has-bad-whatis-entry usr/share/man/man3/log4tango.3.gz
[... cut ...]
I: libtango9: spelling-error-in-binary usr/lib/x86_64-linux-gnu/libtango.so.8.4.3 instanciate instantiate
I: libtango9: spelling-error-in-binary usr/lib/x86_64-linux-gnu/libtango.so.8.4.3 precission precision
W: libtango9: package-name-doesnt-match-sonames libtango8
I: libtango9: no-symbols-control-file usr/lib/x86_64-linux-gnu/libtango.so.8.4.3
W: tango-common: maintainer-script-should-not-use-recursive-chown-or-chmod postinst:67
```

I just put this here in order to have the information somewhere.

Maybe we can solve a bunch of them for the release.

t-b, 2019-11-05:
@picca There is a lot of noise in these warnings like spelling errors etc. Please use also the SONAME fix from the referenced PR.

picca, 2019-11-05:
Hello the easyer for me is to fetch from a tag.

what about tagging an rc2.
We can use plenty of these tags :))

mliszcz, 2019-11-05:
Hi @picca, you can ignore warnings coming from `sphinx-build`. This is known issue, see e.g. build logs on RTD: https://readthedocs.org/projects/tango-controls/builds/9687876/

t-b, 2019-11-05:
@picca 

> I: tango source: wildcard-matches-nothing-in-dep5-copyright lib/cpp/client/zmq.hpp (paragraph at line 15)

We don't ship zmq.hpp anymore.

> I: libtango-doc: font-in-non-font-package usr/share/doc/tango/html/_static/fonts/FontAwesome.otf.gz

sphinx package them by default so I think you sould add lintian overrides for them.

> liblog4tango-doc:

In the future we will not ship liblog4tango as separate library anymore so I would not fix these.

> W: tango source: sphinxdoc-but-no-sphinxdoc-depends
> P: tango source: rules-requires-root-missing

Your turf I guess?

> I: tango-accesscontrol: hardening-no-fortify-functions usr/lib/tango/TangoAccessControl

Is there something I need to fix?

Spelling error fixes:
- https://github.com/tango-controls/cppTango/pull/621
- https://github.com/tango-controls/starter/pull/7
https://github.com/tango-controls/TangoDatabase/pull/20

> P: tango-db: missing-systemd-service-for-init.d-script tango-db

Is that something you can fix?

t-b, 2019-11-05:
@picca You might want to subscribe to https://github.com/tango-controls/TangoSourceDistribution/issues/56.
Against what debian are you building? Unstable? Is buster also feasible?

picca, 2019-11-06:
I target for now experimental, in order to check that it build on all architectures.
then I will upload into unstable.
Then it will be possible to build an official backport if requested by peoples.

Cheers

t-b, 2019-11-06:
@bourtemb I can't request a review for https://github.com/tango-controls/starter/pull/7. Could you have a look and merge it?

t-b, 2019-11-13:
rc2 of the tango source distribution is out.

t-b, 2020-06-11:
rc6 is out, TangoSourceDistribution as well.

picca, 2020-06-16:
Uploaded into Debian experimental right now.

you can test it and gives some feedback.

thanks

picca, 2020-06-16:
https://alioth-lists.debian.net/pipermail/debian-science-maintainers/2020-June/082674.html

[3 more replies]

closed 2022-02-24 by t-b