tango-controls/cppTango #602: appveyor: Add support for automatic uploading of releases to github

t-b, 2019-10-23:
> @bourtemb Regarding "Add Windows installers as assets to the GitHub release": Is that currently done with a script (where is that?) or if not, is it worth having one?

I think it was done manually the last time. Automating this process would be much better of course.

> Appveyor does also support that out of the box. See https://www.appveyor.com/docs/deployment/github/.

Good idea. I think we should try that.

_Originally posted by @bourtemb in https://github.com/tango-controls/cppTango/issues/594#issuecomment-545540317_

t-b, 2020-04-09:
According to @bourtemb this is not working.

t-b, 2020-04-23:
In case the automatic uploading is not working, here is what needs to be done semi-manually:
- Build: https://ci.appveyor.com/project/NexeyaSGara/cpptango-37ylj
- Wait until build is finished
- Go to https://ci.appveyor.com/account/NexeyaSGara/environment/48608/settings (we are deploying to an environment)
- Add the tag you wish to deploy against (can but doesn't have to exist)
- Add a personal access token with `repo` scope
- Deploy each build

t-b, 2020-04-24:
Check if it works in https://github.com/tango-controls/cppTango/releases/tag/9.3.4-rc5 and if yes, forward port the new access token.

bourtemb, 2020-05-14:
>     * [ ]  Check if it works in https://github.com/tango-controls/cppTango/releases/tag/9.3.4-rc5 and if yes, forward port the new access token.

@t-b, did it work or did you use the semi-manual method again?

t-b, 2020-05-14:
Nope I used the semi-manual method again. I need to get that working in a separate test repo. ATM I don't understand why it is not working.

t-b, 2020-06-12:
@bourtemb So I'm getting closer. https://ci.appveyor.com/project/bourtemb/cpptango-tt3w8/branch/tango-9-lts/job/0gbmxfm3i43qo35y did not work as the auth_token was wrong. The error message 

```
Error creating GitHub release: Provider setting not found or it's value is empty. If secure setting is used please check that value was encrypted (or YAML was exported) while being logged under correct account.
Parameter name: auth_token
```

suggests that I can not encrypt a secret for a project running under your account. Can you make me a collaborateur on your appveyor project?

t-b, 2020-06-12:
Or even better move it to a newly created tango-controls orga on appveyor?

bourtemb, 2020-06-12:
I added cppTango team as admin for cppTango project from my appveyor account.
Would that be sufficient for you, @t-b ?
I have no idea how to move it to tango-controls orga appveyor.
I think there are currently at least 3 appveyor accounts which are used for building cppTango PR with appveyor: my account, @NexeyaSGara and @Ingvord 's accounts.

t-b, 2020-06-12:
@bourtemb THanks. That should do the trick for now.

PR #616 (2020-02-28): appveyor.yml: Add inline deployment on tag push
Automate the deployment from appveyor.

We want to publish all artifacts as a new draft prelease on tag push
only.

I've published 9.3.4-rc1 from an environment in appveyor. This is an alternative solution to the inline deployment method.

For playing around I've used an OAuth token from my user. But we can not merge it as it is as using a token from a human user account is unsafe. This OAuth token allows read/write access to all repositories.

I propose to create a machine user which has only access to the cppTango repo and generate an OAuth tok [... cut]
closed 2022-02-24 by t-b