tango-controls/cppTango #603: travis: Only build on new PRs?

t-b, 2019-10-23:
We currently always have two travis invocations per PR because one is for the PR and one for the branch. This is a waste of CI time IMHO and I would disable the building for the branch.

Opinions?

bourtemb, 2019-10-24:
If you build from a branch from your fork instead of creating a branch in tango-controls/cppTango repo, you will see only one travis build when you push commits to the branch of your fork.
If we find a way to avoid the duplicated builds and ensure there is a build after a PR merge in tango-9-lts and 9.3-backports branches, we go for it.

t-b, 2019-10-24:
Got it, thanks.

I don't see a way of restricting the builds only to the important branches while still being able to build all tags. See https://docs.travis-ci.com/user/customizing-the-build/#safelisting-or-blocklisting-branches.

I'll close the bug and encourage fork usage ;)

closed 2019-10-24 by t-b