tango-controls/cppTango #609: Handle deprecated warnings with latest libzmq/cppzmq

t-b, 2019-10-29:
Compiling with latest libzmq, a33f1101 (Merge pull request #3724 from benjdero/qol-improvements, 2019-10-28) and latest cppzmq, fdb2f139 (Merge pull request #356 from phlptp/addon_include_convention, 2019-10-16) , gives the following slew of warnings:

```
zmqeventsupplier.cpp: In constructor ‘Tango::ZmqEventSupplier::ZmqEventSupplier(Tango::Util*)’:
zmqeventsupplier.cpp:249:36: warning: ‘void zmq::message_t::copy(const zmq::message_t*)’ is deprecated: from 4.3.1, use copy taking non-const reference instead [-Wdeprecated-declarations]
     endian_mess_2.copy(&endian_mess);
                                    ^
In file included from ../../../lib/cpp/client/event.h:41:0,
                 from ./tango.h:146,
                 from zmqeventsupplier.cpp:33:
/home/firma/cppzmq/include/zmq.hpp:448:10: note: declared here
     void copy(message_t const *msg_)
          ^~~~
zmqeventsupplier.cpp:254:56: warning: ‘void zmq::message_t::copy(const zmq::message_t*)’ is deprecated: from 4.3.1, use copy taking non-const reference instead [-Wdeprecated-declarations]
     endian_mess_heartbeat_2.copy(&endian_mess_heartbeat);
                                                        ^
In file included from ../../../lib/cpp/client/event.h:41:0,
                 from ./tango.h:146,
                 from zmqeventsupplier.cpp:33:
/home/firma/cppzmq/include/zmq.hpp:448:10: note: declared here
     void copy(message_t const *msg_)
          ^~~~
zmqeventsupplier.cpp:273:52: warning: ‘void zmq::message_t::copy(const zmq::message_t*)’ is deprecated: from 4.3.1, use copy taking non-const reference instead [-Wdeprecated-declarations]
     heartbeat_call_mess_2.copy(&heartbeat_call_mess);
                                                    ^
In file included from ../../../lib/cpp/client/event.h:41:0,
                 from ./tango.h:146,
                 from zmqeventsupplier.cpp:33:
/home/firma/cppzmq/include/zmq.hpp:448:10: note: declared here
     void copy(message_t const *msg_)
          ^~~~
zmqeventsupplier.cpp: In member function ‘virtual void Tango::ZmqEventSupplier::push_heartbeat_event()’:
zmqeventsupplier.cpp:892:63: warning: ‘bool zmq::detail::socket_base::send(zmq::message_t&, int)’ is deprecated: from 4.3.1, use send taking message_t and send_flags [-Wdeprecated-declarations]
                 heartbeat_pub_sock->send(name_mess,ZMQ_SNDMORE);
                                                               ^
In file included from ../../../lib/cpp/client/event.h:41:0,
                 from ./tango.h:146,
                 from zmqeventsupplier.cpp:33:
/home/firma/cppzmq/include/zmq.hpp:1255:10: note: declared here
     bool send(message_t &msg_,
          ^~~~
zmqeventsupplier.cpp:893:75: warning: ‘bool zmq::detail::socket_base::send(zmq::message_t&, int)’ is deprecated: from 4.3.1, use send taking message_t and send_flags [-Wdeprecated-declarations]
                 heartbeat_pub_sock->send(endian_mess_heartbeat,ZMQ_SNDMORE);
                                                                           ^
In file included from ../../../lib/cpp/client/event.h:41:0,
                 from ./tango.h:146,
                 from zmqeventsupplier.cpp:33:
/home/firma/cppzmq/include/zmq.hpp:1255:10: note: declared here
     bool send(message_t &msg_,
          ^~~~
zmqeventsupplier.cpp:895:63: warning: ‘bool zmq::detail::socket_base::send(zmq::message_t&, int)’ is deprecated: from 4.3.1, use send taking message_t and send_flags [-Wdeprecated-declarations]
                 heartbeat_pub_sock->send(heartbeat_call_mess,0);
                                                               ^
In file included from ../../../lib/cpp/client/event.h:41:0,
                 from ./tango.h:146,
                 from zmqeventsupplier.cpp:33:
/home/firma/cppzmq/include/zmq.hpp:1255:10: note: declared here
     bool send(message_t &msg_,
          ^~~~
zmqeventsupplier.cpp:910:48: warning: ‘bool zmq::detail::socket_base::send(zmq::message_t&, int)’ is deprecated: from 4.3.1, use send taking message_t and send_flags [-Wdeprecated-declarations]
                 event_pub_sock->send(dummy_mess);
                                                ^
In file included from ../../../lib/cpp/client/event.h:41:0,
                 from ./tango.h:146,
                 from zmqeventsupplier.cpp:33:
/home/firma/cppzmq/include/zmq.hpp:1255:10: note: declared here
     bool send(message_t &msg_,
          ^~~~
zmqeventsupplier.cpp:917:68: warning: ‘void zmq::message_t::copy(const zmq::message_t*)’ is deprecated: from 4.3.1, use copy taking non-const reference instead [-Wdeprecated-declarations]
                 endian_mess_heartbeat.copy(&endian_mess_heartbeat_2);
[... 200 lines cut ...]
In file included from ../../../lib/cpp/client/event.h:41:0,
                 from ./tango.h:146,
                 from zmqeventsupplier.cpp:33:
/home/firma/cppzmq/include/zmq.hpp:448:10: note: declared here
     void copy(message_t const *msg_)
          ^~~~
zmqeventsupplier.cpp:1520:34: warning: ‘void zmq::message_t::copy(const zmq::message_t*)’ is deprecated: from 4.3.1, use copy taking non-const reference instead [-Wdeprecated-declarations]
   endian_mess.copy(&endian_mess_2);
                                  ^
In file included from ../../../lib/cpp/client/event.h:41:0,
                 from ./tango.h:146,
                 from zmqeventsupplier.cpp:33:
/home/firma/cppzmq/include/zmq.hpp:448:10: note: declared here
     void copy(message_t const *msg_)
          ^~~~
zmqeventsupplier.cpp:1544:35: warning: ‘void zmq::message_t::copy(const zmq::message_t*)’ is deprecated: from 4.3.1, use copy taking non-const reference instead [-Wdeprecated-declarations]
    endian_mess.copy(&endian_mess_2);
                                   ^
In file included from ../../../lib/cpp/client/event.h:41:0,
                 from ./tango.h:146,
                 from zmqeventsupplier.cpp:33:
/home/firma/cppzmq/include/zmq.hpp:448:10: note: declared here
     void copy(message_t const *msg_)
          ^~~~
```

bourtemb, 2019-10-31:
Interesting... and quite annoying... since the methods which are advised as replacement have been introduced at the same time as the deprecation warning, in cppzmq > 4.3.0
So this means that if we want to continue to support cppzmq <= 4.3.0, we will have to handle both cases in the code or we will get this warning, and maybe an error in the future if they decide to release an incompatible version of cppzmq.   :-(
Will they break the compatibility before a new major version of libzmq is released?

t-b, 2020-09-30:
A new version is out https://github.com/zeromq/cppzmq/releases/tag/v4.7.0.

closed 2022-02-24 by t-b