tango-controls/cppTango #622: Change input reference arguments to const references

mliszcz, 2019-11-06:
It is often necessary to create a named temporary variable just to pass it to Tango APIs, because *input* parameters are passed by non-const references.

E.g. instead of writing:
```cpp
ApiUtil.get_db_ind("hostname", 12345)
```
we must write:
```cpp
std::string tmp = "hostname"
ApiUtil.get_db_ind(tmp, 12345)
```

We should mark all input reference parameters as `const`.

Some examples:

- [ ] cppapi/client/ApiUtil.h
    ```
	int     get_db_ind(std::string &host,int port);
	static void device_to_attr(const DeviceAttribute &,AttributeValue &,std::string &);
    ```

- [ ] cppapi/client/Database.h
    ```
	void add_device(DbDevInfo &dev_info);
	DbDevImportInfo import_device(std::string &dev_name);
	void export_device(DbDevExportInfo &info);
	DbDevFullInfo Database::get_device_info(std::string &dev)
	std::string get_class_for_device(std::string &dev_name);
	DbDatum get_class_inheritance_for_device(std::string &dev_name);
	void add_server(std::string &ds_name, DbDevInfos &devs);
	void delete_server(std::string &ds_name);
	...
    ```

- [ ] cppapi/client/DeviceProxy.h
    ```
	virtual CommandInfoList *get_command_config(std::vector<std::string> &cmd_names);
	virtual std::vector<DeviceDataHistory> *command_history(std::string &cmd_name,int depth);
	virtual AttributeInfoList *get_attribute_config(std::vector<std::string> &att_names);
	virtual AttributeInfoListEx *get_attribute_config_ex(std::vector<std::string> &att_names);
	virtual void set_attribute_config(AttributeInfoList &atts);
	virtual void set_attribute_config(AttributeInfoListEx &atts);
	virtual std::vector<DeviceAttribute> *read_attributes(std::vector<std::string> &att_names);
	virtual DeviceAttribute read_attribute(std::string &att_name);
	...
    ```

- [ ] cppapi/client/AttributeProxy.h
    ```
	virtual void write(DeviceAttribute &da);
	...
    ```

- [ ] cppapi/client/DevicePipe.h
    ```
	void set_data_elt_names(std::vector<std::string> &names);
	...
    ```

- [ ] cppapi/client/DeviceData.h
    ```
	void operator << (std::vector<short>&);
	void operator << (std::vector<unsigned short>&);
	void operator << (std::vector<DevLong> &);
	void operator << (std::vector<DevULong> &);
	void operator << (std::vector<DevLong64> &);
	void operator << (std::vector<DevULong64> &);
	void operator << (std::vector<float>&);
	void operator << (std::vector<double>&);
	...
    ```

t-b, 2019-11-06:
We can use [clang-tidy's google-runtime-references](https://clang.llvm.org/extra/clang-tidy/checks/google-runtime-references.html) to find all these and ensure that it stays like that. And of course this will break ABI.

mliszcz, 2019-11-07:
It's not that easy. `google-runtime-references` bans *all* non-const references, i.e:

```cpp
void set_value(std::string& s)
{
    s = "test";
}
```

```
<source>:3:29: warning: non-const reference parameter 's', make it const or use a pointer [google-runtime-references]
void set_value(std::string& s)
                            ^
```

I was unable to find automated check for what we want. We need to go through all functions one by one. Another option is to adopt `google-runtime-references` and pass output parameters as pointers. Or simply return values, but I don't know if we can rely on RVO / copy elision in C++03.

t-b, 2019-11-07:
@mliszcz There is a configurable whitelist for types which are allowed to be passed as non-const reference. In most cases you can rely on NRVO/RVO, I've done some digging a couple of years ago, https://www.byte-physics.de/cpp-copy-elision.html.

closed 2022-02-24 by t-b