tango-controls/cppTango #625: Report C++17 incompatibility to upstream omniORB

t-b, 2019-11-15:
omniORB uses throw specifications and these are not allowed anymore in C++17. More details are in a55ed574 (CMakeLists.txt: Fix compilation problems on debian buster, 2019-08-01).

AFAIR the tango-controls community has a support contract with the omniORB developers. This issue should be raised to them through that channel.

bourtemb, 2019-11-15:
Indeed with have a contract for omniORB support.
Are you sure this has not already been fixed in the latest omniORB release (4.2.3)?

t-b, 2019-11-15:
@bourtemb I've checked the svn repo already. And that is still the case in the most recent version (r6342).

t-b, 2020-04-09:
I'll try to compile cppTango with omniORB 4.3.0 beta 1 and check if it works now with C++17.

bourtemb, 2020-04-09:
omniORB 4.2.4 has been released too this week and should provide a fix for that too.

t-b, 2020-06-12:
omniORB 4.3.0 now compiles with C++17. It is broken with c++20, I wrote an email https://www.omniorb-support.com/pipermail/omniorb-list/2020-June/032130.html.

t-b, 2020-06-14:
And fixed in the svn branches.

closed 2020-07-15 by t-b