tango-controls/cppTango #630: Remove second database instance from tests

mliszcz, 2019-11-27:
Hi all,

PR #608 added a second database instance to the tests to reproduce a scenario where device names from a group are resolved against wrong TANGO_HOST database.

After some testing it looks like it is possible to reproduce the issue with just one database, if TANGO_HOST is not specified (or empty).
* without correction: `tango://host2:port/dev/*` pattern will be resolved to `dev/a/b` which needs TANGO_HOST to be specified
* with the correction, `tango://host2:port/dev/*`  is resolved correctly to `tango://host2:port/dev/a/b`  and the library does not even look for TANGO_HOST variable.

I propose to remove the second database instance to simplify bringing #568 (which needs to start fresh database for each test) up to date.

@bourtemb, @t-b, @Ingvord is it ok for you? If yes, I'll send a PR and remove the second db (TANGO_HOST2).

bourtemb, 2019-11-27:
> I propose to remove the second database instance to simplify bringing #568 (which needs to start fresh database for each test) up to date.
 
Fine by me.

t-b, 2019-11-27:
@mliszcz Sounds good!

t-b, 2019-12-12:
Closed via #630.

closed 2019-12-12 by t-b