tango-controls/cppTango #648: Question: possibility to use boost libraries?

mliszcz, 2019-12-27:
Hi!

I think cppTango codebase could benefit a lot from including some utility library that provides types like: optional, variant, ranges, custom iterators, etc.

While this functionality is gradually being added to C++ standard library (C++17 and 20 releases), Boost project already provides header-only libraries with all these features.

My questions:
* can we add dependency on header-only boost libraries? This of course will impact end-users building from source but packages are available most OSes - I've checked Debian, Arch and CentOS
* if no, do we want to re-implement these features (when needed) in some utility header file in tango, so that we can have `Tango::optional` with API similar to `std::optional` and `boost::optional`?

A few example use-cases:
* indicate possibility of missing value at type level:
  ```cpp
  boost::optional<Command*> DeviceImpl::get_command(CommandName)
  ```
* create joined iterator:
  ```cpp
  boost::any_range<Command*, > DeviceImpl::get_all_commands()
  {
    return boost::join(device_class->commands, this->local_commands)
  }
  ```
* possibility to unify handling of different data types at runtime:
  ```cpp
  boost::variant<DevDouble, DevLong, ...> value;
  auto result = boost::apply_visitor(handle_value_with_overloads, value);
  ```
* and at compile time:
  ```cpp
  boost::variant<...> value = ...
  using AllTypes = boost::mpl::list<DevDouble, DevLong, ...>;
  boost::mpl::for_each<AllTypes>([](auto t) {
    using DataType = decltype(t);
    bool is_correct_type = value.get<DataType*>() != nullptr;
    if (is_correct_type) { ... }
  }
  ```

mliszcz, 2019-12-27:
@bourtemb @t-b @Ingvord what is your opinion?

andygotz, 2020-01-06:
@mliszcz this sounds interesting but needs more discussion of what the implications will be for all platforms. Are there any platforms people are using Tango V9 on which do not have Boost? In the past we avoided such dependancies because of this problem and also because some of the libraries were very heavy. Is Boost lightweight if we use only the headers?

Your examples sound very interesting but we need to discuss the pros and cons.

mliszcz, 2020-01-07:
Thanks Andy for looking into this. I understand your concern regarding adding extra dependencies to Tango. I've opened the issue just to discuss the possibilities of having some recent C++ standard library features without having to upgrade to c++17/2a.

I've checked boost availability on some popular operating systems:
* Debian 8, 9, 10: libboost-dev (just header files),
* Centos 6, 7, 8: boost-devel (headers+libs), 
* Arch: boost (headers+libs)
* Alpine: boost-dev (headers+libs)

Is it heavyweight? Depends on what you mean. Does not need to be installed (just add it to compiler search path with `-I`), is not required at runtime (header-only), but all the includes of all libraries weight ~160MB (including fusion which is 17MB large, phoenix is 19MB, geometry is 13MB and spirit is 11MB).

lorenzopivetta, 2020-01-07:
Hi, another point to consider is which Tango version will be affected. tango-9-lts, as per the decision in the last couple of kernel meeting, already requires some (more) recent compilers...
but 9.3-backports is supposed to compile even without C++11 on platforms where Boost is not available. Moreover, the dependency on PyBoost has just been removed from PyTango in favour of PyBind11...

t-b, 2020-01-07:
While I'm a vidid boost user myself as it indeed has a lot of nice tools to write sweet C++, I'm not sure it is good move at this point. I would say we have a lot of other getting-the-code-in-better-shape tasks which at least I consider more urgent (like https://github.com/tango-controls/cppTango/issues/626, https://github.com/tango-controls/cppTango/issues/622, https://github.com/tango-controls/cppTango/issues/618, https://github.com/tango-controls/cppTango/issues/613, https://github.com/tango-controls/cppTango/issues/586, https://github.com/tango-controls/TangoSourceDistribution/issues/22).

And even if these are done we need to think about if we make the boost classes part of our external API or only use it internally. Things which are part of latter c++ standards (optional, variant) would be nice to be able to use internally. But we should have clear limits which parts of boost are acceptable. And I would not make boost part of our external API as that AFAIU make things more complicated for pytango.

What is nice from boost as well is boost.test, https://www.boost.org/doc/libs/1_72_0/libs/test/doc/html/index.html. We are using that in a couple of our C++ projects.

ltjax, 2020-02-11:
IMO, there are currently better options for most libraries than boost. It is still a pain to handle, e.g. it has a tendency to break on compiler updates. I would advise using the C++17 standard stuff if available and maybe _adopting_ a lightweight variant (e.g. https://github.com/mpark/variant) or optional (e.g. https://github.com/TartanLlama/optional) implementation. Boost is by no means lightweight - the headers are all very interconnected and if you use one, you will most surely drag in heavyweight parts like Boost.MPI.
For unit testing, I'd personally go with catch2 or gtest instead of Boost.Test

closed 2022-02-24 by t-b