tango-controls/cppTango #656: CMake: convert static/shared variant into a build switch

ltjax, 2020-01-15:
As you might know, I've been working on a conan build for cpptango, that as of today works well on Win10/VS2019 and Debian 9 and 10 with cpptango 9.3.3. (You can check it out on https://github.com/softwareschneiderei/conan-cpptango)
The process was not without pain: I currently need 6 patches and some text replacement. Adding support for older VS versions will require more, basically to work around things you have not yet done for VS2019. I'd like to make future iterations a little simpler, if possible. One of the main problems was the "weird" setup of building shared and dynamic libraries with the same CMake configuration. The more common way would be to add a CMake switch like CPPTANGO_SHARED=ON or CPPTANGO_STATIC=OFF instead.
From what I see, building both adds a lot of complexity to the CMake code - basically, every library is added 3 times now, 2x for windows and once for liinux. All this can be simplified/deduplicated using the switch. Also, things like the VS Runtime now longer need to be hard-coded. 
Now you probably have a reason why it is how it is, and I don't want to step on anyone's toes; so I wanted to hear opinions on this before investing the work for a PR.

t-b, 2020-01-23:
@ltjax Sounds good. Especially patches like https://github.com/softwareschneiderei/conan-cpptango/blob/master/disable_runtime_library_overrides.patch should really be not needed.

ltjax, 2020-02-11:
Alright, I'll look into it - let's see if I can scrape the time budget together. My current plan is to just use https://cmake.org/cmake/help/latest/variable/BUILD_SHARED_LIBS.html

NexeyaSGara, 2020-02-12:
Hi,
@ltjax 
The Cmake version used was mainly made by me, and I wasn't a CMake expert( I'm still not).
So the present code was more to have sometinhg working at first.
So I certainly did some things without knowing that it wasn't the right way to do it.

So having a better Cmake file is very welcome. 

As soon as we keep the current packaged files by building the 2 versions, it's ok from my point of view.
And if this permits to have more conan builds, let's do this.

closed 2020-05-28 by t-b