tango-controls/cppTango #667: Prepare 9.3.4-rc4

t-b, 2020-01-23:
- [ ] Merge PRs
  - [ ] https://github.com/tango-controls/cppTango/pull/664
  - [ ] https://github.com/tango-controls/cppTango/pull/665
- [ ] Tag repo with 9.3.4-rc4
- [ ] Upload appveyor windows builds
- [ ] Merge https://github.com/tango-controls/TangoSourceDistribution/pull/71
- [ ] ping picca

t-b, 2020-05-01:
@picca we have  a new release candidate

closed 2020-05-01 by t-b

picca, 2020-05-02:
Uploaded into experimental.

So if you could test it and give me some feed back :)
here the build log

https://buildd.debian.org/status/package.php?p=tango&suite=experimental

there is bunch of warning during the build

cheers

t-b, 2020-05-02:
Thanks! Is https://salsa.debian.org/science-team/tango/-/jobs/711621 the current lintian output?

picca, 2020-05-02:
yes :).