tango-controls/cppTango #675: Archive periodic events no longer received after attribute polling restart

bourtemb, 2020-01-29:
Archive periodic events are no longer sent by the device server if the remote attribute polling is temporary stopped and restarted, archive events based on archive change event thresholds are still sent. It seems like there is no issue with standard PERIODIC events (not archive periodic events).

Here is how you can reproduce the issue with TangoTest device server for instance.

1. Start Tango Test device
2. Set the polling period to 1000 ms on ampli attribute
3. Configure the archive event properties for the ampli attribute with an archive absolute change threshold of 1, and archive period of 3000
4. Start a Tango client (or more) which subscribes to sys/tg_test/1/ampli attribute archive events
5. At this point, the clients receive archive events as expected (every 3s)
6. Stop the polling for ampli attribute
7. The clients receive an API_PollObjNotFound error event
8. Restart the polling for ampli attribute (set to 3000ms for instance)
9. The clients receive 1 archive event for ampli attribute.
10. Wait for some time longer than the archive period event configuration for ampli attribute
11. The archive periodic events are no longer received by the clients/sent by the server after this single event.
12. Set ampli attribute to a different value which should trigger archive change event
13. The clients receive the archive event triggered by the change but the archive periodic events are still not received.

Restarting the device server restores a correct behaviour. Restarting the device sys/tg_test/1 (restart device from jive or DevRestart('sys/tg_test/1') on TangoTest admin device as well as RestartServer command on TangoTest admin device) seems to solve the issue too at some point.

I tested with tango-9-lts branch and 9.3-backports branch, as well with 9.3.3 and the bug seems to be present in all these versions

mliszcz, 2020-04-10:
Hi @bourtemb,
I tried to investigate this issue and attempted to provide a correction in #704 but I'm not sure that I didn't broke anything (I'm getting random crashes in CI in cxx_dserver_misc) and I need help from someone more experienced to clarify the meaning of DELTA_T (see below).

Here is the condition we are using to decide whether we should send archive periodic event from `detect_and_push_archive_event`:
https://github.com/tango-controls/cppTango/blob/7c65021ffdcdea4889978cf91b549c8c32c3ef8d/cppapi/server/eventsupplier.cpp#L637-L641
`ms_since_last_periodic` is the difference between current time and the time we sent the last event:
https://github.com/tango-controls/cppTango/blob/7c65021ffdcdea4889978cf91b549c8c32c3ef8d/cppapi/server/eventsupplier.cpp#L592
Now, in the scenario you describe, `detect_and_push_archive_event` can be called from two paths of execution:
* from `PollThread::poll_attr` which uses following code to calculate `now_ms`:
  https://github.com/tango-controls/cppTango/blob/7c65021ffdcdea4889978cf91b549c8c32c3ef8d/cppapi/server/pollthread.cpp#L1693-L1702
* from `DServer::rem_obj_polling` when you disable polling, which calculates `now_ms` as follows (**note lack of `DELTA_T` adjustment**):
  https://github.com/tango-controls/cppTango/blob/7c65021ffdcdea4889978cf91b549c8c32c3ef8d/cppapi/server/attribute.cpp#L4457-L4463

Example timestamps:
```
polling start:
                    now_ms=584507667118.501953
attr.archive_last_periodic=0.000000
    ms_since_last_periodic=584507667118.501953

polling tick:
                    now_ms=584507668612.118042
attr.archive_last_periodic=584507667118.501953
    ms_since_last_periodic=1493.616089

possibly more ticks ...

disable polling (now_ms is much larger because DELTA_T is not subtracted):
                    now_ms=1586507670645.799072
attr.archive_last_periodic=584507670112.881958
    ms_since_last_periodic=1002000000532.917114

enable polling again (now_ms is ok, delta is negative, always < archive_period):
                    now_ms=584507674148.561035
attr.archive_last_periodic=1586507670645.799072
    ms_since_last_periodic=-1001999996497.238037
```

**Basically, sometimes we use timestamps relative to the Unix epoch and sometime relative to the `DELTA_T`th second of the Unix epoch. This causes the issue.**

I tried to figure out what DELTA_T means but for me it looks like just some random number to get prettier (lower) timestamps. DELTA_T is set to 1002000000 seconds, which translates to `Tuesday, October 2, 2001 7:20:00 AM GMT+02:00 DST`

It was added by @taurel in r8925 in Tango 1.6:
```
------------------------------------------------------------------------
r8925 | taurel | 2002-04-09 16:45:23 +0200 (wto, 09 kwi 2002) | 2 lines

See Tango WEB pages for list of changes

------------------------------------------------------------------------
```

Note that 1.6 was released (2001/10/08) shortly after when Unix timestamp was equal to DELTA_T (October 2, 2001):
```
$ svn diff -c 8925 server/tango_const.h
Index: server/tango_const.h
===================================================================
--- server/tango_const.h   (revision 8924)
+++ server/tango_const.h   (revision 8925)
@@ -11,6 +11,9 @@
 // $Revision$
 //
 // $Log$
+// Revision 1.6  2001/10/08 09:03:14  taurel
+// See tango WEB pages for list of changes
```

bourtemb, 2020-04-14:
@taurel, I know this is very old stuff, but do you recall anything related to this DELTA_T constant?
Do you remember why it was introduced and why it was set to October 2, 2001 7:20:00 AM GMT+02:00 DST ?
Could it be something introduced to avoid overflows when doing some operations on the timestamps?

taurel, 2020-04-14:
As far as I remember, it was to avoid overflow. A kind of anti 2000 year bug

mliszcz, 2020-04-14:
Thanks @taurel and @bourtemb!
I need more time to work on this because one of the tests in cxx_dserver_misc is crashing if I adjust archive periodic event time calculation by DELTA_T or if I set DELTA_T to 0. What I plan to do when I sort this crash out is to extract time calculations to some common function so that we always do it in the same way. Then I'll try to use large enough types (from std::chrono) and get rid of this DELTA_T.

PR #704 (2020-09-24): Correct archive periodic event sending after polling restart
Fixes #675.
PR #778 (2020-11-01): Backport fix 675 no archive periodic events after polling restart
Backport of #704.
Fixes #774. Fixes #675.
closed 2020-09-24 by t-b