tango-controls/cppTango #684: Event data corruption

asenchenko, 2020-02-13:
We have several threads that continuously pushing change event for spectrum attributes. Length of data is about 1000 - 3000. Some events (up to 60%)  contains data from different event.
I think it happens because same push_mutex is used for data protection and for async notification from zmq (tg_unlock function) . Linux implementation of omni_condition uses  pthread_cond_wait which release the associated mutex.  

[image]

debian/ubuntu, Tango 9.2.5a, zmq 4.3.2 ,  4 cores CPU

bourtemb, 2020-03-04:
Many thanks @asenchenko for the detailed analysis and for the Pull Request.
What you are describing seems to make sense and was probably not so easy to find.
We will have a detailed look at #685.
As far as I understood from our e-mail exchanges, you are already using this patched version successfully in your institute since several weeks?
Do you have by any chance a simple code sample showing the issue?
If not, don't worry. What you have done is already impressive and very useful.

bourtemb, 2020-03-05:
@asenchenko , just to understand better your use case:
Do you have several devices exported by your device server?
Are you using the default [serialization model](https://tango-controls.readthedocs.io/en/latest/development/advanced/threading.html#serialization-model-within-a-device-server) (serialization per device)?
Since push_event methods are taking the Tango monitor lock, I think the problem would not occur if the device server would export only one device.
I think it would also not occur if the device server is exporting several devices from the same class and configured to use a serialization by class or by server.
It think it would also not occur if the device server is exporting several devices from different classes and configured to use a serialization by server.

So your PR #685 should make it more robust in the cases which are different than the ones I described above.

asenchenko, 2020-03-05:
> Do you have by any chance a simple code sample showing the issue?

I have code that demonstrate, but this code emulates push_event behavior and contains only zmq related part.

asenchenko, 2020-03-05:
> @asenchenko , just to understand better your use case:
> Do you have several devices exported by your device server?
> Are you using the default [serialization [...]

We have several devices (~10). 
Sorry, i did not provide this information.
Yes, we are using the default serialization model. Data readout is performed asynchronously in dedicated threads. These threads push events with acquired data.