tango-controls/cppTango #706: Build failure cppTango arch: openbsd

delleceste, 2020-04-20:
Hello.
Just trying to build tango on openBSD I get this error:

[ 13%] Building CXX object cppapi/server/CMakeFiles/server_objects.dir/device.cpp.o
In file included from /home/giacomo/devel/cppTango/cppapi/server/device.cpp:37:
In file included from /home/giacomo/devel/cppTango/cppapi/server/tango.h:145:
In file included from /home/giacomo/devel/cppTango/cppapi/client/event.h:40:
/usr/local/include/zmq.hpp:1205:16: warning: 'send' is deprecated: from 4.3.1, use send taking message_t and send_flags [-Wdeprecated-declarations]
        return send(msg, flags_);
               ^
/usr/local/include/zmq.hpp:1189:5: note: 'send' has been explicitly marked deprecated here
    ZMQ_DEPRECATED("from 4.3.1, use send taking message_t and send_flags")
    ^
/usr/local/include/zmq.hpp:41:31: note: expanded from macro 'ZMQ_DEPRECATED'
#define ZMQ_DEPRECATED(msg) [[deprecated(msg)]]
                              ^
In file included from /home/giacomo/devel/cppTango/cppapi/server/device.cpp:37:
In file included from /home/giacomo/devel/cppTango/cppapi/server/tango.h:96:
In file included from /home/giacomo/devel/cppTango/cppapi/client/devasyn.h:31:
/usr/include/c++/v1/map:709:17: error: object of type 'std::__1::pair<const std::__1::basic_string<char> &, const Tango::DevFailed &>' cannot be assigned because its copy
      assignment operator is implicitly deleted
        __ref() = __v.__get_value();
                ^
/usr/include/c++/v1/__tree:1670:35: note: in instantiation of member function 'std::__1::__value_type<const std::__1::basic_string<char>, const Tango::DevFailed>::operator='
      requested here
                __cache->__value_ = *__first;
                                  ^
/usr/include/c++/v1/__tree:1600:9: note: in instantiation of function template specialization 'std::__1::__tree<std::__1::__value_type<const std::__1::basic_string<char>,
      const Tango::DevFailed>, std::__1::__map_value_compare<const std::__1::basic_string<char>, std::__1::__value_type<const std::__1::basic_string<char>, const
      Tango::DevFailed>, std::__1::less<const std::__1::basic_string<char> >, true>, std::__1::allocator<std::__1::__value_type<const std::__1::basic_string<char>, const
      Tango::DevFailed> > >::__assign_multi<std::__1::__tree_const_iterator<std::__1::__value_type<const std::__1::basic_string<char>, const Tango::DevFailed>,
      std::__1::__tree_node<std::__1::__value_type<const std::__1::basic_string<char>, const Tango::DevFailed>, void *> *, long> >' requested here
        __assign_multi(__t.begin(), __t.end());
        ^
/usr/include/c++/v1/map:1016:21: note: in instantiation of member function 'std::__1::__tree<std::__1::__value_type<const std::__1::basic_string<char>, const
      Tango::DevFailed>, std::__1::__map_value_compare<const std::__1::basic_string<char>, std::__1::__value_type<const std::__1::basic_string<char>, const
      Tango::DevFailed>, std::__1::less<const std::__1::basic_string<char> >, true>, std::__1::allocator<std::__1::__value_type<const std::__1::basic_string<char>, const
      Tango::DevFailed> > >::operator=' requested here
            __tree_ = __m.__tree_;
                    ^
/home/giacomo/devel/cppTango/cppapi/server/attribute.h:152:7: note: in instantiation of member function 'std::__1::map<const std::__1::basic_string<char>, const
      Tango::DevFailed, std::__1::less<const std::__1::basic_string<char> >, std::__1::allocator<std::__1::pair<const std::__1::basic_string<char>, const Tango::DevFailed> >
      >::operator=' requested here
class Attribute
      ^
/usr/include/c++/v1/utility:329:5: note: copy assignment operator is implicitly deleted because 'pair<const std::__1::basic_string<char> &, const Tango::DevFailed &>' has a
      user-declared move constructor
    pair(pair&&) = default;
    ^
1 warning and 1 error generated.
*** Error 1 in . (cppapi/server/CMakeFiles/server_objects.dir/build.make:245 'cppapi/server/CMakeFiles/server_objects.dir/device.cpp.o': cd ...)
*** Error 2 in . (CMakeFiles/Makefile2:1319 'cppapi/server/CMakeFiles/server_objects.dir/all')
*** Error 2 in /home/giacomo/devel/cppTango/build (Makefile:163 'all')

Any hint where the problem is?

Giacomo

t-b, 2020-04-20:
Which openbsd,  compiler version and tango version?

I'm not fully understanding the error message but it does look like 

```
void Attribute::add_startup_exception(string prop_name,const DevFailed &except)
{
	startup_exceptions.insert(pair<string,const DevFailed>(prop_name,except));
	check_startup_exceptions = true;
}
```
which is causing the error.

delleceste, 2020-04-21:
Hello!
The problem lies in the definition of startup_exceptions. The value type cannot be const

openbsd# c++ -v
OpenBSD clang version 8.0.1 (tags/RELEASE_801/final) (based on LLVM 8.0.1)
Target: amd64-unknown-openbsd6.6
Thread model: posix
InstalledDir: /usr/bin
openbsd# which c++
/usr/bin/c++

bourtemb, 2020-04-21:
This is also one of the errors reported this week when compiling on Mac OS X:
https://www.tango-controls.org/community/forum/c/general/installation/recipe-to-install-tango-9-from-source-on-macos-x-high-sierra/?page=2#post-4296

delleceste, 2020-04-21:
OK Reynald. I succeeded in compiling Tango on openbsd.
Let me know if interested in the small patches.

Giacomo

bourtemb, 2020-04-21:
Hi @delleceste !
Good news!
Sure, please share your patches. Some of our users will be happy to use them.

delleceste, 2020-04-22:
Hello Reynald!
Please find the attached patch.
Double check it still builds fine on Linux before committing!
As you can see, there's just a very little amount of changes!

Reviewing the patch, I noticed that the section

+                       if(it->first == std::string("event_period") ||
it->first == std::string("archive_period") ||
+                                       it->first ==
std::string("rel_change") || it->first == std::string("abs_change")
+                                       || it->first ==
std::string("archive_rel_change") || it->first ==
std::string("archive_abs_change") )

where explicit std::string conversion is done is not necessary.

Let me know if you have any questions, and when it is available on cppTango
git!

Kind regards

Giacomo

bourtemb, 2020-04-22:
Hi Giacomo (@delleceste) , it looks like you used the reply function from your e-mail editor to reply to a github comment.
This generates automatically a github comment on this page: https://github.com/tango-controls/cppTango/issues/706

But it looks like that when you're using this feature, the attached files do not appear.
Or maybe your filename was not with a github supported file extension?

Could you please try again to submit your patch directly from a comment on github?
You can attach files (You can use the ".txt" file extension) to a comment by drag & drop.

Or the best would be even to create a Pull Request.
Then you would be credited directly in the commit and the Travis CI and appveyor tests would run automatically so we would see immediately whether your patch broke something on Linux or Windows.
If you don't want to bother with that, please provide the patch file and we can create the Pull Request for you.

Thanks for your contribution!

delleceste, 2020-04-27:
Hello Reynald.
While trying to commit on a dedicated branch in order to later create a
pull request I got this:

remote: Permission to tango-controls/cppTango.git denied to delleceste.
fatal: unable to access 'https://github.com/tango-controls/cppTango.git/':
The requested URL returned error: 403

So I will try to attach the patch through a comment.

Thanks, have a good day!

bourtemb, 2020-04-27:
I guess you cannot create a branch on tango-controls/cppTango git repository, but you can always create a fork and work from your fork, and then create a Pull Request from your fork.

delleceste, 2020-04-27:
Hello Reynald.
Here's the patch!
Kind regards!
Giacomo
[tango-openbsd.patch.zip](https://github.com/tango-controls/cppTango/files/4538224/tango-openbsd.patch.zip)

bourtemb, 2020-04-27:
There are 2 important branches: tango-9-lts (future 9.4) and 9.3-backports (branch used for 9.3.x versions like latest 9.3.4).

delleceste, 2020-04-27:
Hi Reynald, for this time, I've already attached the patch to my last
github comment.
Please see if you can correctly find it.
Giacomo

bourtemb, 2020-04-27:
> Hi Reynald, for this time, I've already attached the patch to my last github comment. Please see if you can correctly find it.

Hi Giacomo (@delleceste). Thanks, it worked this time!

PR #711 (2020-05-28): Fix compilation errors on OpenBSD (#706)
Thanks to @delleceste for the patch

Close #706.
closed 2020-05-28 by t-b