tango-controls/cppTango #733: Use CMake built-in support for precompiled headers

mliszcz, 2020-06-26:
Since version 3.16 CMake provides support for defining PCHs for targets.

https://cmake.org/cmake/help/v3.16/command/target_precompile_headers.html#command:target_precompile_headers

We should replace our hand-made implementation with that. One of the reasons is to allow CMake to figure out appropriate `-std` flag without specifying it explicitly as in 65534ede8973c681f3a1be123ebf574674c0a29e (which will break again when we move to C++17).

CMake 3.16 is quite new (November 2019) and we cannot make it mandatory (or can we?), therefore I propose the following way forward:
* drop current PCH implementation
* use PCHs only on CMake `>=3.16`
* upgrade CMake in all CI jobs except one (with minimal required version)

t-b, 2020-07-08:
I don't think we should make a newer cmake mandatory. We are already pushing requirements lately, and having builtin PCH support is not a big gain IMHO.

mliszcz, 2020-07-08:
@t-b how about supporting PCHs only when we use CMake 3.16 or better? -DUSE_PCH will be ignored in other cases. There are at least two things we can gain by moving to builtin PCH support:
* more robust `-std` flag handling (as discussed in #732),
* correction for #734 (seems like we have a bug in our PCH implementation)

Also its always ~100 CMake lines less to maintain.

bourtemb, 2020-07-08:
I would wait a bit (maybe a couple of years) before imposing such a recent version of CMake, unless we realize that there are some other features we really need.
And in any case, this feature (PCH) is mainly there to speed up the compilation. If there is an issue with compiling using the USE_PCH option, the user can disable it and there should be no issue compiling cppTango.

mliszcz, 2020-07-09:
Thanks for your your comments! I'll close the issue in this case. We may reopen in the future when we decide we need.

closed 2020-07-09 by mliszcz