tango-controls/cppTango #734: Winvalid-pch in Debian 10 Release job

mliszcz, 2020-06-26:
PCHs are not picked due to _FORTIFY_SOURCE mismatch. This is probably related to hardening flags recently introduced.

https://travis-ci.org/github/tango-controls/cppTango/jobs/702354291
```
[ 11%] Building CXX object cppapi/client/CMakeFiles/client_objects.dir/dbapi_class.cpp.o

cd /home/tango/src/build/cppapi/client && /usr/bin/c++  -D_FORTIFY_SOURCE=2 -D_REENTRANT -D_TANGO_LIB -I/home/tango/include -I/home/tango/src/cppapi/client -I/home/tango/src/cppapi/client/helpers -I/home/tango/src/cppapi/server -I/home/tango/src/cppapi/server/idl -I/home/tango/src/log4tango/include -I/home/tango/src/build/cppapi/server -I/home/tango/src/build/log4tango/include  -O3 -DNDEBUG   -fstack-protector-strong -fPIC -Winvalid-pch -include /home/tango/src/build/cppapi/server/tango.h -std=c++14 -o CMakeFiles/client_objects.dir/dbapi_class.cpp.o -c /home/tango/src/cppapi/client/dbapi_class.cpp

cc1plus: warning: /home/tango/src/build/cppapi/server/tango.h.gch: not used because `_FORTIFY_SOURCE' is defined [-Winvalid-pch]
```

It should be picked by `get_target_property(pch_compile_defs ${ref_target} COMPILE_DEFINITIONS)` but for some reason it is not.

t-b, 2020-07-08:
Internet [1] suggests that FORTIFY_SOURCE and PCH don't work together.

[1]: https://duckduckgo.com/?q=FORTIFY_SOURCE+PCH&t=ffab&atb=v1-1&ia=web

mliszcz, 2020-07-08:
Can you point to the exact link that states that? Maybe DDG reorders results depending on location, cookies or whatever. The first result is SO discussion with no mention of PCHs and the second one is some discussion at Github where folks just suggested to change source compilation flags instead of changing PCH compilation flags.

The problem is clear if you look at compilation flags. This is the PCH:
```
cd /home/tango/src/build/cppapi/server && /usr/bin/c++ -O3 -DNDEBUG -D_REENTRANT -D_TANGO_LIB -I/home/tango/include -I/home/tango/src/cppapi/client -I/home/tango/src/cppapi/client/helpers -I/home/tango/src/cppapi/server -I/home/tango/src/cppapi/server/idl -I/home/tango/src/log4tango/include -I/home/tango/src/build/cppapi/server -I/home/tango/src/build/log4tango/include -fPIC -x c++-header -std=c++14 -c -o /home/tango/src/build/cppapi/server/tango.h.gch /home/tango/src/cppapi/server/tango.h
```

And this is some source file from cppapi:
```
cd /home/tango/src/build/cppapi/client && /usr/bin/c++  -D_FORTIFY_SOURCE=2 -D_REENTRANT -D_TANGO_LIB -I/home/tango/include -I/home/tango/src/cppapi/client -I/home/tango/src/cppapi/client/helpers -I/home/tango/src/cppapi/server -I/home/tango/src/cppapi/server/idl -I/home/tango/src/log4tango/include -I/home/tango/src/build/cppapi/server -I/home/tango/src/build/log4tango/include  -O3 -DNDEBUG   -fstack-protector-strong -fPIC -Winvalid-pch -include /home/tango/src/build/cppapi/server/tango.h -std=c++14 -o CMakeFiles/client_objects.dir/dbapi_class.cpp.o -c /home/tango/src/cppapi/client/dbapi_class.cpp
```

We need to have the same flags (especially `-D` definitions) in both cases, because otherwise compilation of PCH may produce different intermediate code compared to including the header in a cpp file and compiling that.

I haven't investigated much. Probably the issue is in below code and `_FORTIFY_SOURCE` for some reason is not present in `ref_target`'s compilation flags (this is target like e.g. `server_objects` from which we are copying the flags):
https://github.com/tango-controls/cppTango/blob/37153206dc1430f4830d119050c3b982fbd0fffa/configure/cmake_pch_linux.cmake#L39-L54

mliszcz, 2020-08-06:
@t-b Thanks for the new link. But we had a different issue. We are using add_definitions/add_compile_definitions to set _FORTIFY_SOURCE. And it is not stored in target's COMPILE_DEFINITIONS, but only in directory's property. I corrected this in already in #748.

PR #748 (2020-08-13): Correct PCH compile definitions in Release mode
Use directory COMPILE_DEFINITIONS when building PCH

Compile definitions added using add_compile_definitions are not
automatically stored in COMPILE_DEFINITIONS target property. This
works differently from add_compile_options, which impacts targets
COMPILE_OPTIONS.

We need to check directory property to detect _FORTIFY_SOURCE
in Release mode.

Fixes #734
closed 2020-08-13 by t-b