tango-controls/cppTango #740: Problems when cross compiling for ARM

JeanLucPons, 2020-07-13:
Hello,

I've got several problem when trying to cross compile Tango for ARM platform:
gcc version 4.7.3 (Sourcery CodeBench Lite 2013.05-40) 

My cmake build:
```
[debian9-devel.pons] > cmake .. -DCMAKE_CXX_COMPILER:FILEPATH=/segfs/linux/xilinx_sdk/arm-toolchain/bin/arm-xilinx-linux-gnueabi-gcc -DIDL_BASE=/users/pons/tango/tango-idl/install -DOMNI_BASE=/segfs/tango/ORB/omniORB4.2.1_p1/arm_xilinx -DZMQ_BASE=/segfs/tango/transport/zmq4.0.5/arm_xilinx -DIDL_BASE=/users/pons/tango/tango-idl -DTANGO_JPEG_MMX=OFF -DBUILD_TESTING=OFF -DCMAKE_INSTALL_PREFIX=/segfs/tango/release/arm_xilinx/9.3.4-rc6
System TANGO_HOST=acudebian7:10000
No build type specified - default is Release
Using IDL_BASEIDL_BASE=/users/pons/tango/tango-idl
Using OMNI_BASE=/segfs/tango/ORB/omniORB4.2.1_p1/arm_xilinx
Using ZMQ_BASE=/segfs/tango/transport/zmq4.0.5/arm_xilinx
-- Looking for C++ include zmq.hpp
-- Looking for C++ include zmq.hpp - found
-- Check if zmq::socket has a disconnect method: FALSE
Verifying /segfs/tango/ORB/omniORB4.2.1_p1/arm_xilinx/bin/omniidl
omniidl version 1.0
found GNU compiler ...
setup for release build ...
Remove tango.h, tangoSK.cpp and tangoDynSK.cpp from idl if exist
Generate tango.h, tangoSK.cpp and tangoDynSK.cpp from idl
Using OMNIIDL_PATH=/segfs/tango/ORB/omniORB4.2.1_p1/arm_xilinx/bin/
Using IDL=/users/pons/tango/tango-idl/include
-- OMNIIDL_PATH: /segfs/tango/ORB/omniORB4.2.1_p1/arm_xilinx/bin/
-- CMAKE_CURRENT_BINARY_DIR: /users/pons/tango/cppTango/build/cppapi/server/idl
-- CMAKE_CURRENT_SOURCE_DIR: /users/pons/tango/cppTango/cppapi/server/idl
-- CMAKE_INCLUDE_PATH: 
Applying enhancement /users/pons/tango/cppTango/cppapi/server/idl/EnhanceAttributeValue_4
Applying enhancement /users/pons/tango/cppTango/cppapi/server/idl/EnhanceAttributeValue_5
Applying enhancement /users/pons/tango/cppTango/cppapi/server/idl/EnhanceDevPipeData
-- Configuring done
-- Generating done
-- Build files have been written to: /users/pons/tango/cppTango/build
```

**1st problem**: [log4tango] It seems that cmake does not detect stringstream and that the file Portability.hh does not work when stringstream is not found.

```
                 from /users/pons/tango/cppTango/log4tango/src/Appender.cpp:29:
/users/pons/tango/cppTango/log4tango/include/log4tango/Portability.hh:56:26: error: using typedef-name 'std::ostringstream' after 'class'
```

However, the cross compiler used supports well stringstream and by modifying the config.h by hand, it solved the problem.
log4tango/include/log4tango/config.h 

```
/* define if the compiler has stringstream */
/* #undef LOG4TANGO_HAVE_SSTREAM */
```
changed to:
```
/* define if the compiler has stringstream */
#define LOG4TANGO_HAVE_SSTREAM
```

**2nd problem**: [tango] How to disable the building of shared library ?
I don't have omniORB and zmq compiled in shared, so the compilation of tango SHARED failed and prevent the build process to end properly. To make the build process working, I had to comment in configure/cmake_linux.cmake:

```
#add_library(tango SHARED    $<TARGET_OBJECTS:log4tango_objects>
#                            $<TARGET_OBJECTS:client_objects>
#                            $<TARGET_OBJECTS:idl_objects>
#                            $<TARGET_OBJECTS:jpeg_objects>
#                            $<TARGET_OBJECTS:jpeg_mmx_objects>
#                            $<TARGET_OBJECTS:server_objects>)
#target_link_libraries(tango PUBLIC ${ZMQ_PKG_LIBRARIES} ${OMNIORB_PKG_LIBRARIES} ${OMNICOS_PKG_LIBRARIES} ${OMNIDYN_PKG_LIBRARIES} ${CMAKE_DL_LIBS})
#target_compile_options(tango PRIVATE -fPIC)
#target_include_directories(tango PUBLIC ${ZMQ_PKG_INCLUDE_DIRS} ${OMNIORB_PKG_INCLUDE_DIRS} ${OMNIDYN_PKG_INCLUDE_DIRS})

#target_compile_options(tango PUBLIC ${ZMQ_PKG_CFLAGS_OTHER} ${OMNIORB_PKG_CFLAGS_OTHER} ${OMNICOS_PKG_CFLAGS_OTHER} ${OMNIDYN_PKG_CFLAGS_OTHER})
#target_compile_definitions(tango PUBLIC _REENTRANT)

#set_target_properties(tango PROPERTIES
#                      VERSION ${LIBRARY_VERSION}
#                      SOVERSION ${SO_VERSION})
```

**3rd problem** : [log4tango] test was not disabled in log4tango even when using BUILD_TESTING=OFF
I had to comment the following line in log4tango/CMakeList.txt

```
#if (NOT WIN32)
#   add_subdirectory(tests)
#endif(NOT WIN32)
```

mliszcz, 2020-07-13:
Hi @JeanLucPons 

First, can you specify in the original post or ticket title the version of Tango you are using? Judging from the commandline I guess it is 9.3.4-rc6, right? `-DCMAKE_INSTALL_PREFIX=/segfs/tango/release/arm_xilinx/9.3.4-rc6`. Can you also specify your CMake version?

As for the 1st problem:
* Can you try to compile the check manually with your cross-compiler (without any extra flags):
`arm-xilinx-linux-gnueabi-gcc log4tango/config/check_sstream.cpp`
* Could you check if passing `-DLOG4TANGO_HAVE_SSTREAM=ON` to `cmake` fixes the problem?
* We have https://github.com/tango-controls/cppTango/issues/713 for removing this custom stringstream implementation, but even when it is done, it probably will never get into 9.3 as it is a breaking change.

2nd problem - Please try to build just the static lib (`make tango-static`) and install without depending on the default targets (`make install/fast`). This should help with the 3rd problem as well (tests won't be compiled but CMake still will enter that directory).

JeanLucPons, 2020-07-13:
Hi,
Thanks for the fast answer.

Tango version: 9.3.4-rc6
Cmake version: 3.7.2 (idem with 3.13.2)

```
debian9-devel.pons] > /segfs/linux/xilinx_sdk/arm-toolchain/bin/arm-xilinx-linux-gnueabi-gcc ../log4tango/config/check_sstream.cpp 
/tmp/ccWIl7sB.o: In function `main':
check_sstream.cpp:(.text+0x30): undefined reference to `std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream(std::_Ios_Openmode)'
check_sstream.cpp:(.text+0x48): undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::operator<< <std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*)'
check_sstream.cpp:(.text+0x58): undefined reference to `std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::~basic_stringstream()'
check_sstream.cpp:(.text+0x6c): undefined reference to `std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::~basic_stringstream()'
check_sstream.cpp:(.text+0x74): undefined reference to `__cxa_end_cleanup'
/tmp/ccWIl7sB.o:(.ARM.extab+0x0): undefined reference to `__gxx_personality_v0'
collect2: error: ld returned 1 exit status
```

Yes compiling with -DLOG4TANGO_HAVE_SSTREAM=ON "solved" the first issue.

make tango-static work but the install process failed:

```
debian9-devel.pons] > make install/fast
Install the project...
-- Install configuration: "Release"
CMake Error at cmake_install.cmake:56 (file):
  file INSTALL cannot find
  "/users/pons/tango/cppTango/build/libtango.so.9.3.4".

Makefile:132: recipe for target 'install/fast' failed
make: *** [install/fast] Error 1
```

mliszcz, 2020-07-13:
So, the check for stringstream is not compiling. I guess it is [because you are using C compiler to compile C++ code](https://stackoverflow.com/questions/3081815/c-errors-while-compiling). Do you have something like arm-xilinx-linux-gnueabi-**g++** (instead of gcc) in your compiler toolchain? Could you try use that?

For static-only build I think we need to implement special options at CMake level. If you are working on automated installation, for now you may try to create a dummy so file as a workaround and remove that after install:
```
make tango-static
touch libtango.so.9.3.4 libtango.so.9 libtango.so
make install/fast
rm -f $PREFIX/lib/libtango.so*
```

t-b, 2020-07-13:
3rd problem is solved with https://github.com/tango-controls/cppTango/pull/741. I had that already done in another branch and just pulled it out.

JeanLucPons, 2020-07-13:
Thanks to all :)
1st problem solved by g++ instead of gcc
For the 2nd I keep my work around and wait for a clean SHARED/STATIC selection.
For the third, it would be great if the BUILD_TESTING=OFF disabled also test in log4tango, i didn't check the other branch.

t-b, 2020-08-14:
For 2 we can add the option `BUILD_SHARED_LIBS` from https://cmake.org/cmake/help/latest/variable/BUILD_SHARED_LIBS.html defaulting to true.

t-b, 2020-09-02:
@JeanLucPons In #758 we have a PR which allows to disable dynamic library building. Could you try that? Also do you need to that for upcoming 9.3.4 release or is it enough in tango-9-lts?

bourtemb, 2020-09-08:
> @JeanLucPons In #758 we have a PR which allows to disable dynamic library building. Could you try that? Also do you need to that for upcoming 9.3.4 release or is it enough in tango-9-lts?

According to the comments in #758, @JeanLucPons needs that in 9.3-backports branch, since he is using an old compiler not supporting C++14. So tango-9-lts branch cannot be used in his case.

PR #758 (2020-09-15): Allow turning shared library building off
Close #740.
closed 2020-09-15 by t-b