tango-controls/cppTango #747: Disallow for event subscription and unsubscription using different device proxies

mliszcz, 2020-08-06:
> But we should still be able to use unsubscribe_event explicitly to remove subscription created by other proxies.

From previous comments, both @lorenzopivetta and I are not in favour of allowing proxy 1 to cause proxy 2's subscription to be removed (or vice versa).  In your example above, the first proxy should only be allowed to unsubscribe from id 1, and the second proxy should only be allowed to unsubscribe from id 2.

_Originally posted by @ajoubertza in https://github.com/tango-controls/cppTango/issues/353#issuecomment-669485083_

mliszcz, 2020-08-06:
Note that #746 is a pre-requisite and must be merged first before any work on this PR can be started.

bourtemb, 2020-08-06:
I am not against the behaviour proposed by @ajoubertza and @lorenzopivetta but this will require some refactoring.
Right now, DeviceProxy::unsubscribe_event(int event_id) is simply invoking directly EventConsumer::unsubscribe_event(int event_id) on the EventConsumer instance and the EventConsumer object does not know anything about the DeviceProxy object calling him.

The proposal from this issue is probably the safest and more intuitive approach, even though technically speaking, right now, the only thing needed to unsubscribe for a specific event, is the event id and an access (direct or indirect) to the EventConsumer instance (singleton).
I think the event id returned by subscribe_event() is different at each subscription (even when subscriptions are done from different DeviceProxy objects).

closed 2022-02-24 by t-b