tango-controls/cppTango #765: log4tango: Fix the check for int64_t

t-b, 2020-09-10:
We are currently doing
https://github.com/tango-controls/cppTango/blob/40c05aef2fe4988511ccedfe949889f701b1618f/log4tango/config/config.cmake#L57-L67

 but the presence of int64_t does have nothing to do with the platform or the architecture.

See https://stackoverflow.com/a/48460091

mliszcz, 2020-10-21:
Instead of fixing any checks related to that type I suggest to remove them completely. It is used internally in just one function and can easily be replaced with chrono::steady_clock.

log4tango/src/PatternLayout.cpp
```
    struct MillisSinceEpochComponent : public PatternLayout::PatternComponent {
        virtual void append(std::ostringstream& out, const LoggingEvent& event) {
#ifdef LOG4TANGO_HAVE_INT64_T
            int64_t t = event.timestamp.get_seconds() -
                TimeStamp::get_start_time().get_seconds();
            t *= 1000;
            t += event.timestamp.get_milliseconds() -
                TimeStamp::get_start_time().get_milliseconds();
            
            out << t;
#else
            double t = event.timestamp.get_seconds() -
                TimeStamp::get_start_time().get_seconds();
            t *= 1000;
            t += event.timestamp.get_milliseconds() -
                TimeStamp::get_start_time().get_milliseconds();

            out << std::setiosflags(std::ios::fixed)
                << std::setprecision(0) << t;
#endif
```

t-b, 2020-10-21:
@mliszcz If you can provide a patch I'm happy to include that instead.

mliszcz, 2020-10-21:
@t-b I won't be able to do that in a couple of days for sure. Let's keep what you did in #759, it is fine as well.