tango-controls/cppTango #767: ARM Build

flyingfrog81, 2020-09-11:
#ARM Build of the TANGO Kernel

NRC, in Canada, is using an ARM build of the TANGO Kernel on a Linux embedded system. 
Given the raising interest and number of use cases involving ARM architectures, it would be desirable to start building ARM binaries as part of the Continuous Integration process. 

The goal of this issue is to describe the steps to execute an ARM build of the TANGO kernel. 

##HOWTO

delrizzo, 2020-09-23:
Thank you for the invite to contribute to this conversation, Marco. I will attempt to describe our process here. It has really not been very difficult to compile the TANGO libraries for the aarch64 architecture that we are using. 

Previous to linux kernel 5, we were compiling the kernel within the Angstrom distribution, using the Yocto build system. We are currently compiling linux kernel version 5.4 within the poky distribution, and the version is zeus.

I have created a layer called meta-tango, which contains 4 recipes, one for each of omniorb4, zeromq, tango-idl and tango itself. This layer is incorporated into the poky build, and the recipes are executed during the build process of the kernel. All dependencies (of which there are few) are accounted for within the recipes themselves, and we have ensured that any dependencies required are included in the poky distribution.

I have attached these recipe files for reference to this comment. Note that within the recipes, the source repositories are provided from our internal gitlab instance. This was a measure that we took to streamline the establishment of dev environment locally. Essentially, we have cloned source repositories locally to ensure that our team is building from the same version. Note that I have also attached any patches referred to within the recipes. These patches reside within a 'files' directory in the recipe folder.

In order to incorporate the products of these builds into the resulting poky distribution, the IMAGE_INSTALL environment variable needs to be appended with the names of the recipes.

A zip file containing the current version of this layer for the poky distribution (linux 5.4-zeus) is included here.

[meta-tango-meta-tango-5.4-zeus.tar.gz](https://github.com/tango-controls/cppTango/files/5271715/meta-tango-meta-tango-5.4-zeus.tar.gz)

delrizzo, 2020-09-23:
When it comes to compiling TANGO device servers, we employ cmake with an extensive CMakeLists.txt file that is able to incorporate various base classes and other contributing sections of code as required.

If the developer wishes to cross-compile the device server, they simply set a flag in the top-level CMakeLists.txt file, that includes an additional configuration file. The contents of this file (attached) contains a few 'set' commands which inform cmake as to the location of the cross-compilation toolset, which is currently installed locally for each developer. 

aarch64.cmake:
```set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_SYSTEM_PROCESSOR arm)
set(CMAKE_CROSSCOMPILING true)

set(tools /usr)

set(CMAKE_C_COMPILER ${tools}/bin/aarch64-linux-gnu-gcc)
set(CMAKE_CXX_COMPILER ${tools}/bin/aarch64-linux-gnu-g++)
set(CMAKE_AR ${tools}/bin/aarch64-linux-gnu-ar CACHE INTERNAL "")
set(CMAKE_LINKER ${tools}/bin/aarch64-linux-gnu-ld CACHE INTERNAL "")

set(CMAKE_FIND_ROOT_PATH ${CMAKE_SYSROOT})

set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_PREFIX_PATH ${CMAKE_SYSROOT})
```
Incorporating this cmake config into the build, results in an executable for the device server compiled for the aarch64 architecture, that runs on the ARM embedded processor.

t-b, 2020-09-24:
@delrizzo Thanks for the description and the example files.

Regarding the patches
- musl-fixes.patch: Looks sensible, I'll make a PR
- remove_jpeg_mmx.patch: Not needed anymore, as you can now turn it off with `-DTANGO_JPEG_MMX=OFF`.
- remove_test_suite.patch: Dito `-DBUILD_TESTING=OFF`
- p922_1.diff: Against what is that patch applied? This is huge.

delrizzo, 2020-09-24:
Thank you for that info, Thomas... I will endeavour to make those changes in our build next time I have a moment to do so. In response to your question re: p922_1.diff patch, it is a legacy patch that is not applied to the 9.3.3 build. It was applied to the 9.2.2 build which hasn't been built for a few years. It should be removed.

t-b, 2020-09-24:
> p922_1.diff patch, it is a legacy patch that is not applied to the 9.3.3 build. It was applied to the 9.2.2 build which hasn't been built for a few years. It should be removed.

This is what I hoped for as answer :)

We are also now pretty close to the 9.3.4 release, so there will be a reason to upgrade.

delrizzo, 2020-09-24:
Perfect, I look forward to doing that work to compile 9.3.4!