tango-controls/cppTango #788: Enforce a minimum zeromq version with cmake

t-b, 2020-10-20:
We already do that in the TSD, see https://github.com/tango-controls/TangoSourceDistribution/blob/6d7f13574e8c74ddcb2a11b494236ab671f4db9a/assets/configure.ac#L207.

So we should do the same in cmake. And we should not do it with the help of pkg-config as that does not work on non-linux platforms. And the test should not require running a freshly compiled executable as that does not work with cross-compiling.

mliszcz, 2020-10-21:
Maybe we can just put this requirement into README in a visible place?

t-b, 2020-10-21:
No that does not cut it. I just recently had that issue on debian jessie. If you install `libzmq-dev` (2.2.0) on that platform, you get obscure compile errors.

closed 2022-02-24 by t-b