tango-controls/cppTango #789: Use sonar cache to speed up build

t-b, 2020-10-20:
In https://travis-ci.org/github/tango-controls/cppTango/jobs/737510171 we are hitting the travis limit of 50 min.

Last time I tried that was not successful as it is a bit involved. We are using sonar in docker but the cache has to live outside of docker in travis.

mliszcz, 2020-10-21:
In the job you linked looks like compilation took ~10min and tests ~15min. Do we know why sonar needs extra 25 minutes to process the files?
Maybe dropping sonar and running clang-tidy and gcov manually would be faster? (another benefit is that we do not depend on third-party service and can run locally). Also in such setup we may split the coverage check and static analysis into two CI jobs (because these are really independent things and extra ~15min taken by tests can then be used by linter).

t-b, 2020-11-18:
The sonar cache job also does our coverage analysis. Getting both sonarcloud and coveralls to work again was quite some work, so I'm a bit reluctant to spend more time ont that.

PR #811 (2020-11-19): Travis: Use sonar cache
This should speed up builds.

Close #789.
closed 2020-11-19 by t-b