tango-controls/cppTango #802: Attribute data_format is wrong (at least for DevDouble scalar) if taken from a IDL:Tango/Device_3:1.0 device

delleceste, 2020-11-12:
The following code:

`#include <tango.h>

int main(int argc, char *argv[])
{
    if(argc < 3) {
        printf("tango-format device attribute\n");
        return EXIT_FAILURE;
    }
    try
    {
        Tango::DeviceProxy *dev = new Tango::DeviceProxy(argv[1]);
        std::string att(argv[2]);
        Tango::DeviceAttribute da = dev->read_attribute(att);
        Tango::AttrDataFormat f = da.get_data_format();
        printf("%s/%s data format: %d\n", argv[1], da.get_name().c_str(), f);
    }
    catch(Tango::DevFailed &e)
    {
        printf("\e[1;31merror\e[0m:\n");
        Tango::Except::print_exception(e);
    }
    return EXIT_SUCCESS;
}`

shows that the data format returned by  _ get_data_format _ is 3 if the attribute is a  Scalar DevDouble from a Device_3
while the expected format (0)  is returned for a Device_4 or _5

Is this a known issue?
Is there a workaround?

Thanks

bourtemb, 2020-11-26:
Hi @delleceste ,
When I look at the brief history of Tango that you can get in astor (right click on a host --> Tango Version for Servers, and then Help button at the bottom), I see the following:

Release nb | Date | IDL Release | Main changes
-- | -- | -- | --
5.0 | 01/2005 | 3 | New way to code attribute / ...
-- | -- | -- | --
6.0 | 06/2007 | 3 | omniORB 4.1 / ...
7.0 | 04/2009 | 4 | Device locking / **New way to transfer attribute data** / write_read_attribute call / DevEncoded Attributes / ...

There has been a new way to transfer attribute data introduced in IDL4. Was this change impacting the data format?
Some archaeology digging will be necessary here to understand what changed between IDL3 and IDL4

bourtemb, 2020-11-26:
> Is there a workaround?

I guess the easiest work-around is to upgrade the Device Server to Device_4 or Device_5.

delleceste, 2020-11-26:
Hello Reynald. Thanks for your reply.
I think we will have to migrate the old Device3 impls to the most recent version on the server side.
So nevermind. If you don't recall the possible cause, I wouldn't waste too much time digging into this problem.
Thank you

delleceste, 2020-11-26:
Yes Reynald, we can close the issue I guess!

closed 2020-11-26 by delleceste