tango-controls/cppTango #806: [WIP] Compiling  and running the tests on MacOSX

t-b, 2020-11-17:
- Compile on Mac, see https://git.astron.nl/lofar2.0/tango/-/snippets/9
- Install docker
- Apply patch

```patch
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 500c24b8..420b958c 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -53,7 +53,11 @@ add_subdirectory("log4tango")
 add_subdirectory("cppapi")
 
 if(BUILD_TESTING)
-  if(BUILD_SHARED_LIBS)
+       find_program(DOCKER_BINARY docker)
+       if(NOT DOCKER_BINARY)
+           message(WARNING "Not building the tests, because docker is not present.")
+           SET(BUILD_TESTING OFF)
+  elseif(BUILD_SHARED_LIBS)
     add_subdirectory("cpp_test_suite")
   else()
     message(WARNING "Not building the tests, because that is currently supported only when BUILD_SHARED_LIBS is ON")
diff --git a/cpp_test_suite/environment/run_with_fixture.sh.cmake b/cpp_test_suite/environment/run_with_fixture.sh.cmake
index 1d52699c..41efd5c9 100755
--- a/cpp_test_suite/environment/run_with_fixture.sh.cmake
+++ b/cpp_test_suite/environment/run_with_fixture.sh.cmake
@@ -25,7 +25,7 @@ function teardown {
         exec 2<&-
         "@CMAKE_CURRENT_BINARY_DIR@/kill_server.sh" &>/dev/null || true
         docker stop "$tc_tango_container" "$tc_mysql_container" &>/dev/null || true
-    ) &
+    )
 }
 
 if [[ -z "$TANGO_TEST_CASE_SKIP_FIXTURE" ]]; then
diff --git a/cpp_test_suite/environment/setup_database.sh.cmake b/cpp_test_suite/environment/setup_database.sh.cmake
index fdbe1446..ba36f9ba 100755
--- a/cpp_test_suite/environment/setup_database.sh.cmake
+++ b/cpp_test_suite/environment/setup_database.sh.cmake
@@ -18,6 +18,7 @@ docker run \
 
 docker run \
     --name "$tango_container" \
+    - p 10000:10000 \
     --rm \
     -e TANGO_HOST=127.0.0.1:10000 \
     -e MYSQL_HOST=mysql_db:3306 \
diff --git a/cpp_test_suite/environment/start_server.sh.cmake b/cpp_test_suite/environment/start_server.sh.cmake
index 0bbc117d..087c62a8 100755
--- a/cpp_test_suite/environment/start_server.sh.cmake
+++ b/cpp_test_suite/environment/start_server.sh.cmake
@@ -16,6 +16,8 @@ index="$(find \
 
 echo "Starting $server/$instance"
 
+export DYLD_LIBRARY_PATH="@PROJECT_BINARY_DIR@/${server_path}":$DYLD_LIBRARY_PATH
+
 "@PROJECT_BINARY_DIR@/${server_path}/${server}" "$instance" -v5 \
     &> "${TANGO_TEST_CASE_DIRECTORY}/${server}_${instance}.${index}.out" &
 
```

- Run `make test`

See https://www.tango-controls.org/community/forum/c/general/installation/recipe-to-install-tango-9-from-source-on-macos-x-high-sierra/?page=4

t-b, 2020-11-17:
I'm getting

```
byte-physicss-Mac-Pro:build ci$ "/Users/ci/devel/tango-9.4.0/cppTango/build/cpp_test_suite/new_tests/conf_devtest"     DevTest/test     test/debian8/10     test/debian8/11     test/debian8/12     debian8_alias     debian8_attr_alias     test/fwd_debian8/10     test2/debian8/20
libc++abi.dylib: terminating with uncaught exception of type Tango::ConnectionFailed
```

when I run the tests with the above patch.

beenje, 2020-12-11:
@t-b you can't use the container IP address for `TANGO_HOST="${tango_ipaddr}:10000"`.
As the port is exposed you should use `localhost` or the IP of the host machine.

Using a random port:

```diff
docker run \
     --name "$tango_container" \
+    - p 10000 \
     --rm \

+tango_port="$(docker inspect --format='{{(index (index .NetworkSettings.Ports "10000/tcp") 0).HostPort}}' "$tango_container")"

-export TANGO_HOST="${tango_ipaddr}:10000"
+export TANGO_HOST="localhost:$tango_port"
```

t-b, 2020-12-11:
@beenje I'll give that a shot once I continue this journey here.

beenje, 2020-12-11:
This is the result I got with those changes:

```
90% tests passed, 9 tests failed out of 93

Total Test time (real) = 6340.98 sec

The following tests FAILED:
	  6 - CXX::cxx_dserver_misc (Failed)
	  8 - CXX::cxx_blackbox (Failed)
	  9 - CXX::cxx_class_dev_signal (Failed)
	 10 - CXX::cxx_class_signal (Failed)
	 21 - CXX::cxx_signal (Failed)
	 23 - CXX::cxx_misc (Failed)
	 31 - CXX::cxx_fwd_att (Failed)
	 58 - old_tests::ds_cache (Failed)
	 60 - old_tests::lock (Failed)
```

t-b, 2020-12-11:
@beenje Thanks for trying it out. We'd welcome a PR using GH actions for CI testing cppTango on macOS. We just need not to break the existing tests on linux.

t-b, 2020-12-11:
For the tests it is okay to initially fail, we can do that peu a peu later on.

beenje, 2020-12-15:
I'll try to make a PR when I find some time. Will probably wait for #812 decision.
I have one question about having to set `DYLD_LIBRARY_PATH`.  I understand it's due to: https://github.com/tango-controls/cppTango/blob/fe934f0ccf09980e679e1f87b0bd042573a31a9b/cppapi/server/class_factory.cpp#L156

Do we really need to call `dlopen` on the `exe` and not the lib? I can't find that on Linux.
If it's required, can't we get the full path of the exe directly in the code?

closed 2022-02-24 by t-b