tango-controls/cppTango #814: Support for device level dynamic attributes

HEnquist, 2020-11-19:
In sardana, there are many different things running under the same Class. For example a "Motor" can be a quite different thing depending on which actual controller it uses. This causes problems sometimes, when two different motors define an attribute with the same name but with different properties.

The first device to be instantiated then creates all the attributes without issues, and they all get added to the Motor class. Lets say it has an attribute MaxCurrent, which can bb both read and written. When a second device then starts, it will try to add its attributes, but since the same ones already exists there might be conflicts. Let's imagine this motor also has MaxCurrent, but for this controller it is read only. Then the attribute cannot be added.

The solution in Sardana currently is that when each device starts, it removes any conflicting attribute from the class. This isn't a good solution as it causes other problems for the other devices. 

There appears to be no solution to this on the Sardana level, except the very undesirable one of changing the whole concept of how it uses Classes. 

A very nice solution would be if there was an option to make a new attribute on device level. This exists for commands, where there is an option to add them on device level so they never get added to the class.

https://tango-controls.github.io/cppTango-docs/classTango_1_1DeviceImpl.html#a64f3aafd381cd25bb562cdb4074932d9

The modified add_attribute would then be like this:
```
void Tango::DeviceImpl::add_attribute ( Attr * new_attr, bool device = false ) 	

Add a new attribute to the device attribute list.
Attributes are normally constructed in the DeviceClass::attribute_factory() method. Nevertheless, it is still possible to add a new attribute to a device with this method. Please, note that if you add an attribute to a device at device creation time, this attribute will be added to the device class attribute list. Therefore, all devices belonging to the same class created after this attribute addition will also have this attribute.

Parameters
    new_attr	Pointer to the new attribute to be added to the list. This pointer must point to "heap" allocated memory (or to static memory) and not to "stack" allocated memory
    device	    Set this flag to true if the attribute must be added for only this device Default is false (attribute added for the device class) 

Exceptions
    DevFailed	Click here to read DevFailed exception specification 
```

Adding this "device" flag, and letting each device keep it's own list of device level attributes would be a very welcome addition.

This is a simple device server for reproducing the problem that Sardana faces: https://github.com/reszelaz/tango-dynattrs
It should be run with two devices under the same server instance. It then creates two sets of attributes, where one is conflicting with the other. The result is that the second set isn't created, and that device ends up without it's attributes. See the readme in the repo for details on how to configure it.

HEnquist, 2020-11-19:
I would be willing to work on this, if someone could help me a little to get started.

lorenzopivetta, 2020-11-19:
...mmmhh, maybe I'm missing something, but the proposed solution sounds a bit in conflict with OOP. We'll end up with the same device Class exposing a different interface, or with even the same interface but with *different* behavior.

HEnquist, 2020-11-19:
I see your point, but isn't this already the case since different commands can be defined on device level?

lorenzopivetta, 2020-11-19:
Indeed. I suppose the original purpose being to be able to *extend* the interface, and not modify existing methods. Which is, anyway, somehow borderline if you think OO.

reszelaz, 2020-11-19:
Hi all,
IMHO all the dynamic attributes feature is somehow _borderline_ cause as @lorenzopivetta says we are exposing different interface. But it is very convenient and necessary, I think. 
So, it works well now, unless you try to reuse the same attribute name in a different instance of the same device class within the same DS. If you move the device instance to another DS, no one prevents you to do it. The thing is that we would like to have it within the same Sardana server.

lorenzopivetta, 2020-11-19:
Hi @reszelaz ,
well, yes and no, meaning I've always seen dynamic attributes/commands as a convenient mechanism to populate a device interface, granted that all instances of the same Class expose the same interface, or, at least expose a common set with the same behavior. Something like extending a base class. A scenario where some attribute MaxCurrent is RO in one instance and RW in another is undesirable to me.

HEnquist, 2020-11-20:
Hi @lorenzopivetta, 
Right now there is a mismatch between attributes and commands, in that commands can be defined on device lever while attributes cannot. I can only think of one reason for the device level commands, to make it possible to define the same command name with different in/out types in different instances. Otherwise you can just as well use class level.

I think it would make sense if the same rules applied to attributes and commands here. I think device level attributes would be very useful and not cause any issues in practice. But if that's not allowed then IMHO device level commands should also be removed.

lorenzopivetta, 2020-11-20:
Hi @HEnquist ,
I agree with you the current asymmetry is undesirable. Moreover, please note I'm not stating this cannot be done, just observing some approaches are in disagreement with OOP. Otherwise it is true that these additional mechanisms you can use in a, lets say, "proper" way, or you can completely screw the device model... I think this requires a wider contribution, as it is a proposed change o the model. Suggest to discuss at the next kernel meeting.

taurel, 2020-11-20:
Hello all,

Just a bit of history to understand how things are today.

I fully agree with Lorenzo that  dynamic attributes and dynamic commands 
are strange features for a OO CS !

As a legacy kernel developer, I have always found them "strange". It has 
been implemented first for attribute (and only attribute) and at class

level. Then it appears that in many cases this feature helps a lot to 
solve different case of problems.

We also receive remarks about the implementation

at class level and request to have it at device level but this was not 
implemented. Then, dynamic command was also requested and has been

implemented with the choice to have them at device or class level based 
on the remarks we had and from what exist in Java

(as far as I remember).

Cheers

Emmanuel

andygotz, 2020-11-20:
Hi All,

this is an interesting discussion which we have had in the past and I agree with the "ancients" of the community - the original use case for having dynamic attributes at the CLASS level  has been (over?)used for other use cases. The original use case was a device class which had many attributes (potentially tens to a hundred) which were known when writing the device class or only known at runtime. A typical example of this is a Residual Gas Analyser (RGA) where each element should be an attribute and there are over a hundred potential elements. It was therefore more convenient to define the attributes dynamically for this use case instead of statically. BUT this was done once at CLASS/ creation time and then not modified during the lifetime of the device.

As pointed out this breaks the OO philosophy of ONE class ONE interface for ALL instances. Because it is convenient we  accepted to bend this rule for the kind of use case described above. The CLASS creates the attributes dynamically at class creation time and then does not change them during the lifetime of the class. 

Having the same feature at the DEVICE level is a new use case. It still seems strange to have only one class for all Motors.Why not use multiple classes instead? Or abstract classes? If we make an exception what are the side effects? Would it break binary compatibility? Would it fit in 9.3.x or 9.4? How much work is it?

I agree the dynamic behaviour should be consistent for attributes and commands or if there are differences they should be well documented and motivated. 

Note: the generic tools have issues with the current heavy use of dynamic attributes and commands. They are designed for the classical use cases and not the exotic ones. It would be worse if we allow dynamic attributes at the DEVICE level.

Andy

bourtemb, 2020-11-20:
> Note: the generic tools have issues with the current heavy use of dynamic attributes and commands. They are designed for the classical use cases and not the exotic ones. It would be worse if we allow dynamic attributes at the DEVICE level.

I think the main problem with the current versions of the generic tools is that they are not yet using the InterfaceChange events feature which should be there to handle the dynamic attributes properly.  

Is there anyone using this InterfaceChange events feature by the way?

Personally, I'm all in favour of the changes proposed by @HEnquist. I don't like the current situation which I find misleading but I warn you, if it is decided to go that way, it will probably be not so easy to implement because there are many things to consider (ensure the InterfaceChange events are correctly handled in this use case), test many different use cases and because this changes something which was not foreseen at the start of the project.

I think one could try to implement that in tango-9-lts branch where we don't care about binary compatibility and where it is easier to refactor things and to use more modern C++ features.

We can discuss that during the next Tango Kernel Meeting on Thursday 26th November at 15:00 CET.

t-b, 2020-11-20:
@andygotz I'd say "elder" is the right word (like Gandalf).

taurel, 2020-11-20:
Hello all,

There is something I do not get. As far as I remember I have written 
some times ago (some days, weeks, months, years, decade, century (no, this

one is too much) ago), a HowTo (see 
https://tango-controls.readthedocs.io/en/latest/tutorials-and-howtos/how-tos/how-to-dynamic-attributes-device-class.html)

which explains how using Pogo generated code, you can have dynamic 
attributes at the DEVICE level. Yes Andy, it is available.

Therefore, having this feature also for commands is coherent IMHO. I 
think it is also coherent with what you can do using a Java DS.

Definitively this is not OO but if you do a poll about dynamic 
attributes in the community, my guess is that many people want and use them.

Cheers

Emmanuel

andygotz, 2020-11-20:
Wow Emmanuel you are not so ancient at all - you definitely have a good memory still! I admit I have never used this so I forgot about it. My answer was based on the original use case we discussed when introducing dynamic attributes. You obviously anticipated the needs of Sardana 8-)

Does this mean the problem is in PyTango or is everything there in fact? If it is in PyTango then it should not be a problem to add because the C++ core already supports it.

Andy

HEnquist, 2020-11-20:
I read that howto, and I don't see how that is device level. It looks as it's adding dynamic attributes in the same way as Sardana is doing. It says:
```
1. call the Tango::DeviceImpl::add_attribute() method for each device dynamic attribute.
```
The add_attribute() call will add the attribute to the class, which is exactly the problem.

taurel, 2020-11-20:
Re,

On 20/11/2020 14:55, Henrik Enquist wrote:
> I read that howto, and I don't see how that is device level. It looks
> as it's adding dynamic attributes in the same way as Sardana is doing.
> It says:
>
> |1. call the [...]
As it is written in the HowTo, the example uses Pogo generated code and 
the trick is there. More specifically in the code generated in the

xxxClass::device_factory() method. Have a look at it. I agree, it is a 
trick but it works. Now, I don't know if a similar trick is do-able in 
PyTango but

my guess is that it is

Cheers

Manu

lorenzopivetta, 2020-11-20:
> Wow Emmanuel you are not so ancient at all - you definitely have a good memory still!

Emmanuel, I propose you keep working with us for ten more years ;-)

andygotz, 2020-11-20:
I think I understand the example but I am not sure it fits the use case of sardana. From what I understand from the code in the example in the how to the dynamic attribute of the class needs to know about all the dynamic attributes which could be supported. If we take the example Motor then it would mean Motor has to do the a test on all the possible attributes which could exist and then call the correct one. This implies pre-knowledge of all possible dynamic attributes in all Motor classes. Which is not so dynamic. It is indeed a trick but I am not sure it is as dynamic as the dynamic attributes for commands. It seems to me to be a class dynamic attribute which is calling different getters and setters depending on the attribute name which is different for each device.

In order to have a true device dynamic attribute we need to have a list of attributes per device as well as per class. But I presume this is not the case which is why one cannot do what Drew wants.

Correct me if I have misunderstood the example.

Andy

taurel, 2020-11-20:
Re,

On 20/11/2020 15:30, Andy Gotz wrote:
> I think I understand the example but I am not sure it fits the use
> case of sardana. From what I understand from the code in the example
> in the how to the dynamic attribute of the class needs to [...]
In your class, Pogo generates a method which is named 
"add_dynamic_attributes()" in which you code the 
DeviceImpl::add_dynamic_attribute()

using methods add_XXX_dynamic_attribute() also created by Pogo. You do 
that for dynamic attributes that are required for your device

(based on dev properties in the HowTo). This is per device no? And where 
are the tests you are talking about?

I do not really get the point of your last message.

What is sure is that the dynamic attributes has to be defined in Pogo 
(as dynamic attribute)

PS: Note that Pogo also generates a method named 
"add_dynamic_commands()" to be coherent between cmds and atts.

I have already used both  and for me it works and it is by device (once 
more it is horrible to write this knowing we claim Tango

is OO CS !!!)

Cheers

Manu

reszelaz, 2020-11-20:
Hi all,
First, many thanks to all of you for spending your time in this discussion!
Please remember that we are talking about dynamic attributes with the **same name**. For example one device could have attribute `attr1` (READ and DevDouble) and the other device could have attribute `attr1` (READ_WRITE and DevLong). Pogo prevents you already to define them saying "Attribute "attr1" Already Exists !".
If you try to do it programmatically, you get an exception (try the example from the issue description):
```
~/workspace/tango-dynattrs (main)> python3 buggyds.py zreszela
Create dynamic attribute: FirstValue
Create dynamic attribute: SecondValue
Create dynamic attribute: ThirdValue
Create dynamic attribute: FirstValue
DevFailed[
DevError[
    desc = Device test/zreszela/buggyds-02 -> Attribute FirstValue already exists for your device class but with other definition
           (data type, data format or data write type)
  origin = DeviceImpl::add_attribute
  reason = API_AttrNotFound
severity = ERR]
]
Ready to accept request
```

Now my humble observation about the OO CS principles. If we think about the whole Tango CS instance (and not one Tango DS) we already allows that. It is enough to move the device to another process and the device **of the same class** is successfully exported. And I think that one of the Tango principles is that it does not matter in which process the device resides. The only thing which prevents us in having these two devices in the same Tango DS process is the library implementation.

bourtemb, 2020-12-03:
I created the following repo to illustrate and reproduce with a C++ device server the current limitation reported in this issue: https://github.com/bourtemb/DynaTest

Please feel free to play with it.

Please also note that we encountered as well a similar limitation with the dynamic forwarded attributes features a while ago (https://github.com/tango-controls/cppTango/pull/342#issuecomment-278335835), when you have a device server exporting several devices and these devices are creating dynamic forwarded attributes with the same name and pointing to root attributes having different types or read/write types.

closed 2022-02-24 by t-b