tango-controls/cppTango #816: Limit of six device classes in a device server?

DrewDevereux, 2020-11-24:
After experiencing "unidentifiable C++ exception" when trying to add a device to pytango's `DeviceTestContext`, I dug into the code, and eventually found that a device server can have any number of device instances of **up to six device classes**. But as soon as a seventh device class is added, the device server cannot run. The following python code illustrates this problem. The device server runs successfully if `include_seventh_device` is `False`. If set to `True`, then the device server fails to run.
```
from tango import Database
from tango.server import Device, run
from tango.test_context import get_host_ip

class A(Device): pass
class B(Device): pass
class C(Device): pass
class D(Device): pass
class E(Device): pass
class F(Device): pass
class G(Device): pass

# Write device info to text file database
with open("db_file", "w") as f:
    f.write("Debug/debug/DEVICE/A:test/a/1\n")
    f.write("Debug/debug/DEVICE/B:test/b/1\n")
    f.write("Debug/debug/DEVICE/C:test/c/1\n")
    f.write("Debug/debug/DEVICE/D:test/d/1\n")
    f.write("Debug/debug/DEVICE/E:test/e/1\n")
    f.write("Debug/debug/DEVICE/F:test/f/1\n")
    f.write("Debug/debug/DEVICE/G:test/g/1\n")

include_seventh_device = False

# run server
run(
    [A, B, C, D, E, F] + [G]*include_seventh_device,
    args=[
        "Debug",
        "debug",
        "-ORBendPoint",
        f"giop:tcp:{get_host_ip()}:57061",
        "-file=db_file"
    ]
)
```

bourtemb, 2020-11-24:
> 
> 
> I wild guess would be that `i` -> `j` in this line.

I would replace 
`(*data_out)[index] = Tango::string_dup((*data_in)[i + 1].in());index++;`
with 
`(*data_out)[index] = Tango::string_dup((*data_in)[j].in());index++;`

as it is done in FileDatabase::DbGetDeviceProperty  so `i+1` -> `j`

bourtemb, 2020-11-24:
If we refer to DbGetClassProperty command in the Tango Database server, here are the argin and argout expected arguments:

```
Argin description:
Str[0] = Tango class
Str[1] = Property name
Str[2] = Property name
```

```
Argout description:
Str[0] = Tango class
Str[1] = Property number
Str[2] = Property name
Str[3] = Property value number (array case)
Str[4] = Property value
Str[n] = Propery value (array case)
....
```

bourtemb, 2020-11-24:
If I understand well the code, at the line
https://github.com/tango-controls/cppTango/blob/ef0c2be4a3223cbdbcd9521fc66eb8d7363cca87/cppapi/client/filedatabase.cpp#L1647
 , we are in the case where a property name provided as argin does not exist in the class provided in Argin[0].

If we refer to the equivalent code in the TangoDatabase device server, I would say that https://github.com/tango-controls/cppTango/blob/ef0c2be4a3223cbdbcd9521fc66eb8d7363cca87/cppapi/client/filedatabase.cpp#L1647 
corresponds to https://github.com/tango-controls/TangoDatabase/blob/Database-Release-5.16/DataBase.cpp#L2853

So here we want to put the name of the property provided by the user that we didn't find in the specified class.
So `(*data_in)[j].in()` in this case, which is the name we were comparing with the different class properties of the wanted class at line 1622 just above: https://github.com/tango-controls/cppTango/blob/ef0c2be4a3223cbdbcd9521fc66eb8d7363cca87/cppapi/client/filedatabase.cpp#L1622

closed 2022-02-24 by t-b