tango-controls/cppTango #819: Gitlab Migration: Please log into GL with your GH account ASAP

bourtemb, 2020-12-18:
Dear cppTango participant (via commits, comments on issue or pull requests, issue creator),

If you have contributed in any way (commit, issue or Pull Request creation or comment) to cppTango, please log into Gitlab.com using your Github account **ASAP**.
See https://github.com/tango-controls/TangoTickets/issues/47 for details.

For convenience, here is the list of contributors we could already find:  
- [ ] @AntoineDupre 
- [ ] @CorF4
- [ ] @DmitryKuk
- [ ] @FPO28
- [ ] @HEnquist
- [ ] @HFRibeiro
- [ ] @Ingvord 
- [ ] @JeanLucPons 
- [ ] @NexeyaSGara 
- [ ] @abogani 
- [ ] @ajoubertza 
- [ ] @alexElettra
- [ ] @andygotz 
- [ ] @asenchenko
- [ ] @aureocarneiro
- [ ] @beenje
- [ ] @chedburgh 
- [ ] @claudio-scafuri
- [ ] @cmft
- [ ] @delleceste
- [ ] @delrizzo
- [ ] @egorbe
- [ ] @elkinvg
- [ ] @grm84 
- [ ] @gscalamera 
- [ ] @guifrecuni 
- [ ] @hardion
- [ ] @jairomoldes 
- [ ] @jkotan
- [ ] @josefcibulka
- [ ] @juandesant
- [ ] @liujin001
- [ ] @lorenzopivetta 
- [ ] @ltjax
- [ ] @lzytniak
- [ ] @mguijarr
- [ ] @mike-a-smith
- [ ] @mkomel
- [ ] @mliszcz
- [ ] @nleclercq
- [ ] @picca
- [ ] @reszelaz
- [ ] @sergirubio
- [ ] @stanislaw55
- [ ] @t-b
- [ ] @taurel
- [ ] @tiagocoutinho
- [ ] @tjuerges
- [ ] @udyni
- [ ] @vxgmichel

Please tick the box corresponding to your account above when you've already done the login to Gitlab.com using your Github account.
If you are not allowed to tick this box, please simply comment this issue once you've login to gitlab.com using your Github account.

Please note that login on gitlab.com using your Github account is safe and doesn't give your Github password to Gitlab.

Thanks for your cooperation.

asenchenko, 2020-12-18:
Done.

beenje, 2020-12-18:
I have the same username on GH and GL (@beenje) and I checked that all the e-mails I used on GitHub are linked to my GitLab account as well.

abogani, 2020-12-18:
I have done.

bourtemb, 2020-12-18:
> I have the same username on GH and GL (@beenje) and I checked that all the e-mails I used on GitHub are linked to my GitLab account as well.

Thanks @beenje. As far as I understood, if you don't use (at least once) the Github button on gitlab.com to login, you need to ensure your e-mail primary address on Github is also PUBLIC. If it's not the case, the migration tool might not be able to match your commits and comments.
See https://docs.gitlab.com/ee/user/project/import/github.html

> When issues and pull requests are being imported, the importer attempts to find their GitHub authors and assignees in the database of the GitLab instance (note that pull requests are called “merge requests” in GitLab).
For this association to succeed, each GitHub author and assignee in the repository must meet one of the following conditions prior to the import:
    Have previously logged in to a GitLab account using the GitHub icon.
    Have a GitHub account with a publicly visible primary email address on their profile that matches their GitLab account’s email address.

beenje, 2020-12-18:
Thanks @bourtemb. I had missed that my public email was not set in GH. It's now updated.

andygotz, 2020-12-18:
I could not login with github the first time. So I logged in to gitlab with my email, then connected my GL account to GH in my profile. After that I could log into GL with GH. I don't know if others encountered this problem.

jkotan, 2020-12-18:
I have done.

gscalamera, 2020-12-18:
Done

cmft, 2020-12-18:
Done

claudio-scafuri, 2020-12-18:
Done
Cheers

On venerdì 18 dicembre 2020 13:52:05 CET Reynald Bourtembourg wrote:
> Dear cppTango participant (via commits, comments on issue or pull requests, issue creator),
>
> If you have contributed in any way (commit, issue or Pull Request creation or comment) to cppTango, [...]

-- 
Claudio Scafuri

Control Systems
Information Technology Group

Elettra - Sincrotrone Trieste S.C.p.A.
S.S. 14 - km 163,5 in AREA Science Park
IT-34149 Basovizza, Trieste (IT)

claudio.scafuri@elettra.eu

Phone:  +39 040 3758009
Mobile: +39 3357002786

****
The  information  in  this  message  and/or  attachments  is  intended  
solely  for  the  attention  and  use  of  the named addressee and may be 
confidential. If you are not the intended recipient, please notify the sender 
and delete the message and/or attachments without copying or disclosing.Please 
“think before you print”, consider the environment and print it only if 
necessary.

Le informazioni contenute in questo messaggio e/o negli allegati sono 
di pertinenza ed uso esclusivo del destinatario e possono essere confidenziali. 
Se non siete l'esatto destinatario, siete pregati di notificarlo al mittente e 
di cancellare il messaggio e/o gli allegati senza copiarli né diffonderli.Si 
prega di “riflettere prima di stampare”, considerare l’ambiente e poi stampare 
solo se necessario.
****

mkomel, 2020-12-18:
Done

josefcibulka, 2020-12-18:
Done

picca, 2020-12-18:
Done (I hope :)

ltjax, 2020-12-18:
Done

tjuerges, 2020-12-19:
Done.
Note:  I did the same that Andy did.

HEnquist, 2020-12-19:
Done!

Den lör 19 dec. 2020 kl 04:04 skrev Thomas Juerges <notifications@github.com

aureocarneiro, 2020-12-21:
Done!

alexElettra, 2020-12-21:
Done, thanks.

sergirubio, 2020-12-21:
done

bourtemb, 2021-01-13:
Gentle reminder, if your Github account name is listed below, please sign in on Gitlab.com using the sign in with Github button.
If you already have a Gitlab account, you should link your account with your Github account, go to https://gitlab.com/-/profile/account and click on connect to GitHub.
Please add your Github account name in the following Framapad when it is done: https://mensuel.framapad.org/p/migration_ready_2548763689

Here are the accounts who haven't done it yet (or didn't report yet that they had done it):
- @AntoineDupre
- @CorF4
- @DmitryKuk
- @FPO28
- @HFRibeiro
- @JeanLucPons
- @NexeyaSGara
- @chedburgh
- @delrizzo
- @egorbe
- @elkinvg
- @grm84
- @jairomoldes
- @juandesant
- @liujin001
- @lzytniak
- @mguijarr
- @mike-a-smith
- @stanislaw55
- @taurel
- @tiagocoutinho
- @udyni
- @vxgmichel

HFRibeiro, 2021-01-14:
Hi Reynald,

Sorry for not responding the first time.

All the SKA members should have this done already since we work with 
gitlab for a while now. (Mine HFRibeiro)

To help you with this those are the remaining users I can't find on gitlab:

  * @CorF4 <https://github.com/CorF4>
  * @FPO28 <https://github.com/FPO28>
  * @JeanLucPon <https://github.com/JeanLucPons>
  * @NexeyaSGara <https://github.com/NexeyaSGara>
  * @chedburgh <https://github.com/chedburgh>
  * @delrizzo <https://github.com/delrizzo>
  * @liujin001 <https://github.com/liujin001>
  * @mike-a-smith <https://github.com/mike-a-smith>
  * @stanislaw55 <https://github.com/stanislaw55>

Best regards,

bourtemb, 2021-01-14:
@HFRibeiro , thank you very much for your comment.
Having an account on Gitlab.com is not enough for the migration to go fully smoothly.
@cpascual did a first migration test 1 month ago with cppTango.
In your case, we can see that the issue you created on github was not perfectly re-created as created by your Gitlab user:
https://gitlab.com/tango-controls/test_gh_import/cppTango/-/issues/660.
The migrated issue does not look too bad but still, could be better if it was created as if your gitlab.com user would have created it.

So, it is important to associate the gitlab.com account with the github.com account (go to https://gitlab.com/-/profile/account and click on connect to GitHub) and I guess most of the SKA developers didn't have to do that up to now.

HFRibeiro, 2021-01-14:
Thank you Reynal,

Indeed my account was now disconnected from the github, really strange 
because I'm 100% sure I created the gitlab one using the github.

This was when SKA moved also from github to gitlab. I guess something 
changed and we need to manually reconnect again.

Thank you once again.

bourtemb, 2021-01-14:
The cppTango gitlab migration is foreseen on Thursday 21st January 2021.
So if you're mentioned above, please associate your Gitlab account to your GitHub account before that date or connect on Gitlab.com using the GitHub sign in button before that date.
Thanks in advance.

liujin001, 2021-01-26:
Done, maybe too late

elkinvg, 2021-01-26:
> Gentle reminder, if your Github account name is listed below, please sign in on Gitlab.com using the sign in with Github button.
> If you already have a Gitlab account, you should link your account [...]

Sorry.
I added earlier and wrote my Github account name in Framapad

jairomoldes, 2021-01-26:
I already did this on 18-01 and reported to framedata, but I am still listed as not yet done.

bourtemb, 2021-01-28:
I did a import test to Gitlab this morning.
The result can be seen there: https://gitlab.com/tango-controls/test_gh_import/preview-import/cppTango

What is interesting is to see this issue after the migration: https://gitlab.com/tango-controls/test_gh_import/preview-import/cppTango/-/issues/819

Then, we can see all the accounts which have been correctly recognized, and the others.
So, as of today, the following Github accounts have not been recognized correctly on Gitlab:
- @CorF4
- @FPO28
- @JeanLucPons 
- @NexeyaSGara 
- @chedburgh 
- @cmft 
- @delrizzo
- @guifrecuni 
- @mike-a-smith 
- @stanislaw55

@cmft and @guifrecuni , could you please check again because you said you followed the instructions but you're still not recognized on Gitlab.
Please read carefully the instructions on https://github.com/tango-controls/TangoTickets/issues/47
If you already have a github account, it is important to associate the gitlab.com account with the github.com account (you should go to https://gitlab.com/-/profile/account and click on connect to GitHub).

closed 2022-02-24 by t-b