tango-controls/cppTango #826: Conda build problems for 9.3-backports on windows

t-b, 2021-01-15:
It compiled but linking failed:
LINK : fatal error LNK1181: cannot open input file 'msvcstub.lib'

Where does this lib comes from?
set(OMNIORB_PKG_LIBRARIES_DYN "omniORB4_rt.lib;omniDynamic4_rt.lib;omnithread_rt.lib;COS4_rt.lib;msvcstub.lib")

Is it supposed to come with omniORB?
It looks like it was removed in tango-9-lts.

/Benjamin

﻿On 2021-01-15, 15:12, "Thomas Braun" <thomas.braun@byte-physics.de> wrote:

    Am 1/15/2021 um 2:16 PM schrieb Benjamin Bertrand:

    Hi Benjamin,

    the mentioned PR has already been merged. So if you use 0abc6ccc (Merge 
    pull request #824 from tango-controls/no-sed-backport, 2021-01-15) 
    instead of the 9.3.4 tag it should already work.

    Thomas

    > Hi Thomas,
    > 
    > Thanks. Indeed, no sed as there is no Cygwin.
    > Looks like it should be easy to apply the change from this commit as a patch.
    > I'll give it a try and let you know.
    > 
    > /Benjamin
    > 
    > On 2021-01-14, 23:43, "Thomas Braun" <thomas.braun@byte-physics.de> wrote:
    > 
    >      Am 1/14/2021 um 10:36 PM schrieb Benjamin Bertrand:
    > 
    >      Hi Benjamin,
    > 
    >      thx.
    > 
    >      The error from
    > 
    >      ...
    > 
    >      %SRC_DIR%\cppapi\server\device_3.cpp(926): error C2039:
    >      'set_attr_mutex': is not a member of 'Tango::AttributeValue_5'
    >      %SRC_DIR%\cppapi\server\idl/tango.h(3395): note: see declaration of
    >      'Tango::AttributeValue_5'
    >      %SRC_DIR%\cppapi\server\device_3.cpp(928): error C2039:
    >      'set_attr_mutex': is not a member of 'Tango::AttributeValue_5'
    >      %SRC_DIR%\cppapi\server\idl/tango.h(3395): note: see declaration of
    >      'Tango::AttributeValue_5'
    >      %SRC_DIR%\cppapi\server\device_3.cpp(940): error C2039:
    >      'set_attr_mutex': is not a member of 'Tango::AttributeValue_4'
    >      %SRC_DIR%\cppapi\server\idl/tango.h(3363): note: see declaration of
    >      'Tango::AttributeValue_4'
    >      %SRC_DIR%\cppapi\server\device_3.cpp(942): error C2039:
    >      'set_attr_mutex': is not a member of 'Tango::AttributeValue_4'
    >      %SRC_DIR%\cppapi\server\idl/tango.h(3363): note: see declaration of
    >      'Tango::AttributeValue_4'
    >      %SRC_DIR%\cppapi\server\device_3.cpp(959): error C2039:
    >      'rel_attr_mutex': is not a member of 'Tango::AttributeValue_5'
    >      %SRC_DIR%\cppapi\server\idl/tango.h(3395): note: see declaration of
    >      'Tango::AttributeValue_5'
    >      %SRC_DIR%\cppapi\server\device_3.cpp(968): error C2039:
    >      'rel_attr_mutex': is not a member of 'Tango::AttributeValue_4'
    >      %SRC_DIR%\cppapi\server\idl/tango.h(3363): note: see declaration of
    >      'Tango::AttributeValue_4'
    > 
    >      ...
    > 
    >      means that patching the code generated from omniidl was not done. The
    >      code at
    >      https://github.com/tango-controls/cppTango/blob/f4b4b652a90ffd82d2ecf7bb4da79a0f5d783187/cppapi/server/idl/CMakeLists.txt#L31
    >      should do that. I would guess that sed is missing.
    > 
    >      At some point we had a PR which fixed that in tango-9-lts, see
    >      https://github.com/tango-controls/cppTango/pull/638, but looks like this
    >      was not backported to 9.3-backports.
    > 
    >      Thomas
    > 
    >      > Hi!
    >      >
    >      > Please find attached the windows failed conda build log.
    >      >
    >      > I’ll try to push my branch somewhere if you want to try locally.
    >      >
    >      > /Benjamin
    >      >
    > 
    > 
    >      --
    >      () byte physics
    >          Thomas Braun
    >          Schwarzastraße 9
    >          12055 Berlin
    >          Germany
    > 
    >      Web:       www.byte-physics.de
    >      Telephone: +49 (0)30/52 68 23 22
    >      Fax:       +49 (0)30/52 68 43 26
    >      USt-IdNr:  DE276717536
    > 

    -- 
    () byte physics
        Thomas Braun
        Schwarzastraße 9
        12055 Berlin
        Germany

    Web:       www.byte-physics.de
    Telephone: +49 (0)30/52 68 23 22
    Fax:       +49 (0)30/52 68 43 26
    USt-IdNr:  DE276717536

t-b, 2021-01-15:
I think you can just remove msvcstubd.lib as done in 4791f0dd (Rework static/shared variants into a configure-time build-switch (#688), 2020-05-28).

beenje, 2021-01-18:
I removed `msvcstubd.lib` but I also need to set `TANGO_INSTALL_DEPENDENCIES=OFF` when compiling with conda. I see that this option was introduced in the same PR. Probably have to backport the whole commit.

t-b, 2021-01-19:
@beenje I've tried that in https://github.com/tango-controls/cppTango/pull/827. It had some heavy conflicts though.

t-b, 2021-01-21:
@beenje Does #827 fix the build problems on windows?

beenje, 2021-01-21:
Yes, I confirm I could build using commit 6f3ca599c3e621966fc9b08a7fb07f54f30f4bab

closed 2022-02-24 by t-b